lgsvl/simulator #1895: Simulation Error: Value of name '_SpawnRate' was not found

pmode, 2021-11-29:
I try to build simulation from source and start simulating. But there are many error messages in the developing console, ` Value of name '_SpawnRate' was not found`, could anyone help me with this issue?
OS: Ubuntu 20.04
Unity Editor Version: 2020.3.3f1
Simulator branch: release-2020.3

EricBoiseLGSVL, 2021-11-30:
please post the player log to see if there are other errors first. Spawn rate is from the tire spray, so a vehicle maybe setup incorrectly
[image]

pmode, 2021-11-30:
I couldn't find any error before this in player.log. I copy binary files which I build from source and replace them to the release version, issue disappears. So I think the git repository might not contain one or more configuation files.

pmode, 2021-11-30:
for more imformation: I create a simple simulation in wise.svlsimulator.com, and ego I choose is Lincoln2017MKZ--Apollo 5.0 (full analysis), bridge is CyberRT.

EricBoiseLGSVL, 2021-12-01:
What exactly are you copying from the binary?