lgsvl/simulator #1899: Radar does not detect NPCs with waypoints

pantonante, 2021-12-01:
Hi, when running a scenario created with the Visual Scenario Editor containing NPCs following waypoints, the radar does not detect the vehicles.
If I create a scenario with a static vehicle, the radar works fine. The radar works fine also with the Random Traffic template.

I get the same behavior with SVL version 2021.1/2.2/3.

EricBoiseLGSVL, 2021-12-02:
We have fixes that we are pushing soon to master to address this

pantonante, 2021-12-02:
That's great! Thank you!

pantonante, 2021-12-07:
@EricBoiseLGSVL do you have an estimate on when this will happen? I'm working on a project where this feature is very important and I would like to plan around it.

EricBoiseLGSVL, 2021-12-08:
We are hoping to push changes to master within the next few weeks.

EricBoiseLGSVL, 2022-01-24:
Latest radar sensor.cs
```
/**
 * Copyright (c) 2019-2021 LG Electronics, Inc.
 *
 * This software contains code licensed as described in LICENSE.
 *
 */

using System.Linq;
using System.Collections.Generic;
using Simulator.Bridge;
using Simulator.Bridge.Data;
using Simulator.Utilities;
using UnityEngine;
using UnityEngine.AI;
using Simulator.Sensors.UI;

namespace Simulator.Sensors
{
    using System;

    [SensorType("Radar", new[] { typeof(DetectedRadarObjectData) })]
    public class RadarSensor : SensorBase
    {
        [SensorParameter]
        [Range(1.0f, 100f)]
        public float Frequency = 13.4f;
        public LayerMask RadarBlockers;

        private List<RadarMesh> radars = new List<RadarMesh>();
        private WireframeBoxes WireframeBoxes;

        private uint seqId;
        private float nextPublish;

        private BridgeInstance Bridge;
        private Publisher<DetectedRadarObjectData> Publish;
        private Rigidbody SelfRigidBody;

        /// <summary>
        /// Dictionary of all agents in the range and their semaphore locks
        /// Semaphore is required because radar can use multiple trigger colliders
        /// Agent has to be cached until any trigger containing it is in the range
        /// </summary>
        private Dictionary<ITriggerAgent, int> AgentsInRange = new Dictionary<ITriggerAgent, int>();
        private Dictionary<ITriggerAgent, DetectedRadarObject> Detected = new Dictionary<ITriggerAgent, DetectedRadarObject>();
        private Dictionary<ITriggerAgent, Box> Visualized = new Dictionary<ITriggerAgent, Box>();

        [AnalysisMeasurement(MeasurementType.Count)]
        public int MaxTracked = -1;
        
        public override SensorDistributionType DistributionType => SensorDistributionType.MainOrClient;
        public override float PerformanceLoad { get; } = 0.2f;
        
        struct Box
        {
            public Vector3 Size;
            public Color Color;
        }

        private void Awake()
        {
            radars.AddRange(GetComponentsInChildren<RadarMesh>());
            foreach (var radar in radars)
            {
                radar.Init();
            }
            SimulatorManager.Instance.NPCManager.RegisterDespawnCallback(OnExitRange);
        }
[... 345 lines cut ...]
                Graphics.DrawMesh(radar.RadarMeshFilter.sharedMesh, transform.localToWorldMatrix, radar.RadarMeshRenderer.sharedMaterial, LayerMask.NameToLayer("Sensor"));
            }
        }

        public override void OnVisualizeToggle(bool state) {}
    }
}

```
This may help if the sensor changes don't solve the issue.  It was the change needed for GT2D detection of add agent in api mode.
PedestrianManager.cs (NPCManager.cs might need the same changes.)
```
298 SimulatorManager.Instance.UpdateSegmentationColors(pedController.gameObject, pedController.GTID);
before
return pedController

and 
322
// Segmentation ID does not change on re-pool, so segmentation colors don't have to be updated
// Just make sure GTID is tracked again after re-spawning from pool
SimulatorManager.Instance.SegmentationIdMapping.AddOrGetSegmentationId(CurrentPooledPeds[i].gameObject, CurrentPooledPeds[i].GTID);

before
foreach (var callback in SpawnCallbacks)
```

pantonante, 2022-01-25:
Is it possible that you also changed the `ITriggerAgent` interface and the `SimulatorManager`?

I'm getting these errors when I try to compile:

```
'ITriggerAgent' does not contain a definition for 'AgentGameObject' and no accessible extension method 'AgentGameObject' accepting a first argument of type 'ITriggerAgent' could be found (are you missing a using directive or an assembly reference?)
'ITriggerAgent' does not contain a definition for 'AgentBounds' and no accessible extension method 'AgentBounds' accepting a first argument of type 'ITriggerAgent' could be found (are you missing a using directive or an assembly reference?)
'SimulatorManager' does not contain a definition for 'TriggersManager' and no accessible extension method 'TriggersManager' accepting a first argument of type 'SimulatorManager' could be found (are you missing a using directive or an assembly reference?)
```

EricBoiseLGSVL, 2022-01-25:
Yes, since our current repo is ahead of latest release we have quite a few changes in simulator that will be missing.

pantonante, 2022-01-26:
Any chance you can share these changes in some branch or here?

EricBoiseLGSVL, 2022-01-26:
Sorry, with simulator being sunset we will not be pushing any changes.  With the current state internally, it will cause too many issues if we share or push latest.

alperenbayraktar, 2022-03-01:
@pantonante did you manage to send the radar data after all?

pantonante, 2022-03-02:
No, I decided to generate the radar from ground truth data outside SVL because it was easier in my project. You can read the code here https://github.com/pantonante/apollo/blob/master/verification/bridge/fake_radar.py

alperenbayraktar, 2022-03-02:
I see, deleting the isTrigger check at the start of the WhileInRange method also fixes the problem.