lgsvl/simulator #1900: Seeking help in designing a truck-trailer vehicle - Importing from Unity to LGSVL

Bmoradi93, 2021-12-03:
Hello lgsvl developers,
My MSc research area is designing local planners for truck-trailer vehicle.
I have done my simulation in Gazebo. But I need to migrate to LGSVL. It helps me to integrate my research with Autoware.Auto.
I'm looking for someone to help me in this direction.
We need a truck-trailer system in LGSVL compatible with ROS2 or Autoeare.Auto.
We need all the sensor and hitch angle topics in ROS2 foxy. 
Here is a relevant video [tutorial](https://www.youtube.com/watch?v=R_Z65kimCII)

GooVincent, 2022-01-15:
Please how to watch the parameters of the vehicle, like Lincoln2017MKZ? Maybe the mass or RPM info etc.

EricBoiseLGSVL, 2022-01-17:
@GooVincent Jag parameters are the same as the Lincoln2017.  Just clone the open source repo and look at the rigidbody settings for mass in Unity Editor.  RPM is available from sensors.