lgsvl/simulator #1902: Bug Report

yumianhuli2, 2021-12-03:
1 When I checkout the branch to release-2021.3(unity 2020.3.3f1 from web doc),there is a bug when I opened the project from Unity
[image]

2 The unity version of master branch is unity 2020.3.19f1, Can I use it for what docker version of apollo?
Thanks!
.

EricBoiseLGSVL, 2021-12-03:
We don't have a crash when opening release-2021.3 with unity 2020.3.3f1
Please post more information on what hardware, OS, GPU and installation of Unity you have.
Do not use master it is not stable.  Did you open simulator release with unity 2020.3.19?  If so you may have broken assets.  Unity is not backwards compatible.

yumianhuli2, 2021-12-03:
> simulator

OS:ubuntu18.04
GPU:NVIDIA RTX 1070
open unity version:2020.3.3f1
simulator branch:release-2021.3

yumianhuli2, 2021-12-03:
> We don't have a crash when opening release-2021.3 with unity 2020.3.3f1 Please post more information on what hardware, OS, GPU and installation of Unity you have. Do not use master it is not stable. [...]

OS:ubuntu18.04
GPU:NVIDIA RTX 1070
open unity version:2020.3.3f1
simulator branch:release-2021.3

yumianhuli2, 2021-12-03:
> We don't have a crash when opening release-2021.3 with unity 2020.3.3f1 Please post more information on what hardware, OS, GPU and installation of Unity you have. Do not use master it is not stable. [...]

nvidia-smi
Sat Dec  4 07:45:17 2021       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 460.80       Driver Version: 460.80       CUDA Version: 11.2     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  GeForce GTX 1070    Off  | 00000000:01:00.0  On |                  N/A |
| 46%   23C    P8     8W / 200W |    455MiB /  8116MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      2294      G   /usr/lib/xorg/Xorg                 18MiB |
|    0   N/A  N/A      2469      G   /usr/bin/gnome-shell               72MiB |
|    0   N/A  N/A      3147      G   /usr/lib/xorg/Xorg                123MiB |
|    0   N/A  N/A      3281      G   /usr/bin/gnome-shell               85MiB |
|    0   N/A  N/A      3666      G   /usr/lib/firefox/firefox          148MiB |
+-----------------------------------------------------------------------------+

EricBoiseLGSVL, 2021-12-04:
Are you running this on a remote machine? Can you open a blank project?

EricBoiseLGSVL, 2021-12-04:
Do you have mesa drivers installed?

EricBoiseLGSVL, 2021-12-04:
Do you have a player log from the editor to see why Unity Editor is crashing?

yumianhuli2, 2021-12-04:
> Are you running this on a remote machine? Can you open a blank project?

1 no
2 yes

yumianhuli2, 2021-12-04:
> Do you have mesa drivers installed?

Do you mean vulkan?
Yes, I have use sudo apt-get install libvulkan1

yumianhuli2, 2021-12-04:
> Do you have a player log from the editor to see why Unity Editor is crashing?
[Player.log](https://github.com/lgsvl/simulator/files/7653504/Player.log) is my log
Thank you!

EricBoiseLGSVL, 2021-12-06:
I don't see anything in the log that helps.  Did you delete the old assets in the appdata folder since you are updating to a new version?

yumianhuli2, 2021-12-07:
> I don't see anything in the log that helps. Did you delete the old assets in the appdata folder since you are updating to a new version?

The first time an error is reported, the second time can be entered, but the box containing the Build button does not show any assets

EricBoiseLGSVL, 2021-12-07:
Do you have asset repos cloned to the Assets/External/ folders?

yumianhuli2, 2021-12-07:
> 

> Do you have asset repos cloned to the Assets/External/ folders?

1、Do you mean that it doesn't matter if you get an error the first time, but you can finally enter the editor ？
2、Ok, I've read the chapter on Building Assets.All Assets can be downloaded in the https://content.lgsvlsimulator.com/maps, then I need to download to the Simulator which map/Assets/External/subfolder?

EricBoiseLGSVL, 2021-12-07:
1. Yes, as long as there are no errors when you have the editor open.  It often take Unity Editor a few restarts to fully update.
2. No, these links state that they are for version 2020.06.  Github has all the public repos to clone, e.g. Assets/External/Environments/RepoName or Assets/External/Vehicles/RepoName
3. Be sure to enable git lfs when cloning and make sure you are on the correct branch/tags on those repos as well as simulator core.

sixiaixixiya, 2022-08-25:
> Are you running this on a remote machine? Can you open a blank project?

[image]

EricBoiseLGSVL, 2022-08-26:
Are you getting a playerlog to see what the errors are?