lgsvl/simulator #1906: Inside driving view

yuchendoudou, 2021-12-08:
Hello lgsvl developers, 
https://www.jianshu.com/p/d314f70b26ba said svl has a realistic in-car driving environment, and they changed their view to be in-car. However I can't change ours like them, We only have FreeCamera, CinematicCamera, DriverViewCamera and FollowCamera, none of them has a in-car view. The version we used is release-2021.3. So, how can I do to change our view to be in-car, and use G923 to control the wheel in-car.

EricBoiseLGSVL, 2021-12-10:
You can change the driverview camera to be inside the vehicle (we didn't because some vehicles don't have finished models/textures inside).  Build the bundle and you will have the driver camera change.
Then use the WheelControlSensor.  If the controls don't work because the sensor is made for the Logitech G920, then you will need to make a new sensor.  You will need to edit the control inputs.  You need to install the driver software, open in the editor, open the input debug logger so you can press the buttons to see what the inputs are.  Build the sensor and rename it in the webui.

yuchendoudou, 2021-12-10:
> You can change the driverview camera to be inside the vehicle (we didn't because some vehicles don't have finished models/textures inside). Build the bundle and you will have the driver camera [...]

Thank you for your answer, I have made a new sensor for G923, and it worked well. But after I changed the function CreateDriverViewTransform() :
[image]
The result like this:
[image]

Is this right?

Besides, we also want to achieve force feedback for G923, so can you give me some advice.
Thank you

EricBoiseLGSVL, 2021-12-10:
You can move the existing game object in the vehicle DriverView to the exact position or change this code, up to you.
Force feedback requires you to access the steering wheel sdk.  You can look at our older simulator before we moved to HDRP to see how we did this [here](https://github.com/lgsvl/simulator-2019.05-obsolete).

yuchendoudou, 2021-12-11:
Thank you for your replay, I will try it.

closed 2021-12-11 by yuchendoudou