lgsvl/simulator #1908: How to export GPS coordinates or shpfile of any element in opendrive（.xodr） in unity？

Bardbo, 2021-12-09:
**How to export GPS coordinates or shpfile of any element in opendrive（.xodr） in unity？**
For example, the longitude and latitude of the road in CubeTown.
In the screenshot，I can only see local positions.
In addition, my Unity cannot display the rendered scene, only the road lines，why？
Thanks（Excuse me for bad English！I got it with Baidu translation）
[image]

EricBoiseLGSVL, 2021-12-10:
No worries I think I got ya.
Sorry, we do not support .shp files.
Look to the exporter code in OpenDriveMapExporter.cs. We show in the editor only the local but we calculate the world positions on export.  We don't show the world because of excess serialization issues.
Did you use git lfs when you cloned the cubetown repo?  Do you have the light on in the Scene view?
[image]

Bardbo, 2021-12-14:
Thank you for your answers.
Yes, git lfs can solve this problem~
But I still don't know how to export latitude and longitude using c# script.
I think I should learn about unity and c#.
Incidentally, does this function have a python API？
Thank you very much~

EricBoiseLGSVL, 2021-12-14:
In MapOrigin we have lat and long calculations multiple ways, see if you can add to the exporter code.  Let us know what you find out and good luck.