lgsvl/simulator #1911: How to build the simulator entirely offline?

jim18324, 2021-12-09:
Hi all, 
I was wondering is there a way to build the simulator entirely offline without accessing the internet. Just like using Developer debug mode and Development settings in the editor. I had set up the scene and vehicle, and they work well. I want to make it one step further, build it as a binary without using the internet, but I have no clue. 

Any suggestion and help? Thank you!

EricBoiseLGSVL, 2021-12-10:
No sorry, this is not available at this time.  You will need to alter the source code to do this.  Unity needs to validate the Unity editor and account you use to build the binary, they will not have a legal workaround for this purpose.

EricBoiseLGSVL, 2021-12-10:
No, we require internet connection to run the simulation the first time.  After that, you can run offline simulations.

closed 2022-01-10 by jim18324