lgsvl/simulator #1913: Error when starting unity

bilibilicharger, 2021-12-09:
Hello, I encountered this problem during startup.
[image]
 I used to use 2020.3.10 Unity to start the simulator, but later deleted the original simulator file and downloaded a simulator again. As a result, this problem occurred.
[image]
[image]

bilibilicharger, 2021-12-09:
Also prompted this
[image]

EricBoiseLGSVL, 2021-12-10:
Please look to our documentation on cloning simulator [here](https://www.svlsimulator.com/docs/installation-guide/build-instructions/).
1. Install Unity 2020.3.3f1
2. Checkout the release-2021.3 of simulator source.
3. Clone asset repos that you need in Assets/External/

Safemode means you have errors and cannot open the project.  Do the above steps with a new project and please use git lfs for the simulator and all asset repos.  Let us know how you do after that.

bilibilicharger, 2021-12-10:
I did follow the tutorial in the first two steps, but I haven't done the third step yet. I'll try to clone the simulator again.

bilibilicharger, 2021-12-10:
Because gitbash on Windows cannot clone simulator, I use Ubuntu command line clone simulator 2021 3. Then use it on windows. Will this affect it?

bilibilicharger, 2021-12-10:
I think I have solved this problem, but there is a bug in the sta
[image]
rtup times. I don't know if there is a problem.

EricBoiseLGSVL, 2021-12-10:
What branch are you on for simulator? Did you use gitlfs install and gitlfs pull?

fyl000, 2021-12-16:
> Please look to our documentation on cloning simulator [here](https://www.svlsimulator.com/docs/installation-guide/build-instructions/).
>
> 1. Install Unity 2020.3.3f1
> 2. Checkout the release-2021.3 of [...]

I encountered a similar  problem during Unity startup. And I have followed these 3 steps exactly as you showed.
After I entered Unity in SafeMode, the error is :
[image]
I don't know what I should do now...

EricBoiseLGSVL, 2021-12-16:
You have added speed tree to the unity editor and it is throwing errors. Remove this from the package list in unity package manager

bilibilicharger, 2021-12-16:
@EricBoiseLGSVL 
Sorry, I forgot to tell you that I have solved this problem. I don't use LFS and  clone the latest version of simulator directly

fyl000, 2021-12-16:
> You have added speed tree to the unity editor and it is throwing errors. Remove this from the package list in unity package manager

Thanks for your reply. 
After I git clone simulator 2021.3 in Ubuntu which has installed LFS, and copied the files to Windows, I solved the problem.
I hope this might help other beginners :)