lgsvl/simulator #1915: 3D Ground Truth Sensor can detect vehicles but not pedestrians (2021.3)

AIasd, 2021-12-11:
Hi,

I am using 2021.3 and Apollo6.0(modular testing) and encounter an issue of 3D Ground Truth Sensor only being able to detect vehicles but not pedestrians. (I followed the suggestion in #1886 and already set the position of the sensor to (0,0,10).)

[image]

AIasd, 2021-12-11:
The pedestrian is generated through Python API so I guess the cause might be similar to #1797 except this is for 3D Ground Truth Sensor rather than 2D Ground Truth Sensor

closed 2021-12-14 by AIasd

JasonBourne1998, 2022-05-17:
> The pedestrian is generated through Python API so I guess the cause might be similar to #1797 except this is for 3D Ground Truth Sensor rather than 2D Ground Truth Sensor

Hi @AIasd , I have also encountered this problem but this is a sunset simulator so it could not be solved by official, do you have a corresponding solution to this problem?