lgsvl/simulator #1917: New ego vehicle's asset bundle upload fails

eslaaam3, 2021-12-13:
I'm trying to build a new ego vehicle, and I have been experimenting for about a week now

Here is what I did:

1. Getting the simulator source
    Steps:
    ```
    $ git lfs install
    ```
    Output:
    ```
    Git LFS initialized.
    ```
    ```
    $ git clone https://github.com/lgsvl/simulator.git
    $ cd simulator/
    ```
    ```
    $ git checkout 2021.2.2
    ```
    ```
    $ git lfs ls-files
    ```
    Output:
    ```
    55c5d08704 * Assets/Animations/HumanoidIdle.fbx
    ceab209964 * Assets/Animations/HumanoidWalk.fbx
    ff13e179c1 * Assets/Animations/HumanoidWalkTurn.fbx
    83864de466 * Assets/Editor/MapUI/MapUIBoundryCurb.png
    8826c97edd * Assets/Editor/MapUI/MapUIBoundryDotWhite.png
    5bb3edb17b * Assets/Editor/MapUI/MapUIBoundryDotYellow.png
    e4263a5202 * Assets/Editor/MapUI/MapUIBoundryDoubleWhite.png
    d73670f7d7 * Assets/Editor/MapUI/MapUIBoundryDoubleYellow.png
    17de035fb0 * Assets/Editor/MapUI/MapUIBoundrySolidWhite.png
    cd05455ea4 * Assets/Editor/MapUI/MapUIBoundrySolidYellow.png
    6bedc3c7e4 * Assets/Editor/MapUI/MapUIBoundryUnknown.png
    52bbdb49cc * Assets/Editor/MapUI/MapUIBoundryUnknownPro.png
    be628006f6 * Assets/Editor/MapUI/MapUIBoundryVirtual.png
    c4491d0aaf * Assets/Editor/MapUI/MapUICurved.png
    3c4b44f45c * Assets/Editor/MapUI/MapUICurvedPro.png
    edee661df0 * Assets/Editor/MapUI/MapUIDelete.png
    49d0a86e3b * Assets/Editor/MapUI/MapUIDeletePro.png
    3a70a165d6 * Assets/Editor/MapUI/MapUISignalBack.png
    c81931bb16 * Assets/Editor/MapUI/MapUISignalBackPro.png
    d589fff590 * Assets/Editor/MapUI/MapUISignalDown.png
    c20f00475b * Assets/Editor/MapUI/MapUISignalDownPro.png
    3bf975623c * Assets/Editor/MapUI/MapUISignalForward.png
    ccb0ccded7 * Assets/Editor/MapUI/MapUISignalForwardPro.png
    f7a7f415a2 * Assets/Editor/MapUI/MapUISignalHorizontal2.png
    ba416facc1 * Assets/Editor/MapUI/MapUISignalHorizontal2Pro.png
    24521d0eb2 * Assets/Editor/MapUI/MapUISignalHorizontal3.png
    2e80b0d403 * Assets/Editor/MapUI/MapUISignalHorizontal3Pro.png
    721303c245 * Assets/Editor/MapUI/MapUISignalSingle.png
    66cdca79a8 * Assets/Editor/MapUI/MapUISignalSinglePro.png
    88caab7b61 * Assets/Editor/MapUI/MapUISignalUp.png
    1c8d36a002 * Assets/Editor/MapUI/MapUISignalUpPro.png
    4bd16cf7ad * Assets/Editor/MapUI/MapUISignalVertical2.png
    9f86ebce45 * Assets/Editor/MapUI/MapUISignalVertical2Pro.png
    4bc15fb32f * Assets/Editor/MapUI/MapUISignalVertical3.png
    db0160de5f * Assets/Editor/MapUI/MapUISignalVertical3Pro.png
    90d4477963 * Assets/Editor/MapUI/MapUIStoplineLeft.png
    c3d37bd366 * Assets/Editor/MapUI/MapUIStoplineLeftPro.png
    f8409a661b * Assets/Editor/MapUI/MapUIStoplineRight.png
    ff0e94c9a7 * Assets/Editor/MapUI/MapUIStoplineRightPro.png
    9e1ab4864f * Assets/Editor/MapUI/MapUIStraight.png
    7bad54caa2 * Assets/Editor/MapUI/MapUIStraightPro.png
    e64fa5f954 * Assets/Editor/MapUI/MapUIWaypoint.png
    90f3f99da7 * Assets/Editor/MapUI/MapUIWaypointPro.png
    d12e83d4d6 * Assets/Effects/Materials/RainParticleTexture.png
    7cfd9535ab * Assets/Effects/Materials/RainParticleTexture02.png
[... 380 lines cut ...]

5. Importing ego vehicle (FBX)
    Pivot point:
    [image]
    Prefab hierachy:
    [image]
    Building AssetBundle
    [image]
    Console output:
    [image]
    [image]

6. Upload AssetBundle to [https://wise.svlsimulator.com/vehicles/profile/add](https://wise.svlsimulator.com/vehicles/profile/add)
    Selecting the file by clicking the drag and drop area throws this error
    [image]
    And dragging and dropping directly throws this one
    [image]

My machine
OS: Windows 10
GPU: Nvidia GTX 1080Ti 12GB
CPU: Core i7-5960X
RAM: 16GB

Any help is appreciated.

EricBoiseLGSVL, 2021-12-14:
OK, thanks for the images, it helps.  Looks like the install is good but something went wrong with the bundle build:
[image]

Can you screen shot the M2 folders you cloned to Assets/External/Vehicles/M2 please?  Also, show what scripts you attached to the vehicle.

Last thing, why are you not using the latest simulator release?  2021.3 has many fixes that might help.  It requires the same Unity version 2020.3.3f1.

eslaaam3, 2021-12-14:
Hi @EricBoiseLGSVL,

**M2 directory**:
[image]
[image]
[image]

```
$ tree.com //a //f M2
D:\<path to simulator>\SIMULATOR\ASSETS\EXTERNAL\VEHICLES\M2
|   M2.prefab
|   M2.prefab.meta
|   Models.meta
|
\---Models
    |   M2.fbx
    |   M2.fbx.meta
    |   Materials.meta
    |
    \---Materials
            01-Default.001.mat
            01-Default.001.mat.meta
            Black.mat
            Black.mat.meta
            blue_police_light.mat
            blue_police_light.mat.meta
            body_black.mat
            body_black.mat.meta
            body_grey.mat
            body_grey.mat.meta
            LiDAR_grey.mat
            LiDAR_grey.mat.meta
            M2HeadLightHighBeamCookie.png
            M2HeadLightHighBeamCookie.png.meta
            M2HeadLightLowBeamCookie.png
            M2HeadLightLowBeamCookie.png.meta
            Material.mat
            Material.mat.meta
            Metal_Bling_chrome.mat
            Metal_Bling_chrome.mat.meta
            Metal_rough.mat
            Metal_rough.mat.meta
            red_police_light.mat
            red_police_light.mat.meta
            white_police_light.mat
            white_police_light.mat.meta
```

I tried version 2021.3 but it fails in the middle of AssetBundle building process, so I kept going back in time and using older releases one by one and until version 2021.2, the same problem persists when I try to upload and even though the building process takes a while and doesn't fail, I always find the built AssetBundle's size 0 KB.

I hope this helps.

EricBoiseLGSVL, 2021-12-14:
It is odd because you should have no issues with a windows version and 2021.3 is much better for building bundles.  What version of windows are you on?
All components look right and folder structure is right.  Do you run unity hub in admin mode?

eslaaam3, 2021-12-15:
**Windows version**:
[image]

**Cloning version 2021.3**:
```
$ mkdir simulator2021.3
$ cd simulator2021
$ git clone https://github.com/lgsvl/simulator.git --branch 2021.3 --single-branch
$ cd simulator
$ git lfs ls-files
```
Outputs all files preceded with an asterisk, just as my first comment.

**Importing simulator version 2021.3**:
I opened UnityHub as an administrator this time, it throws these warnings/errors in the console on importing:
[image]
[image]

**Building the simulator**:
Console:
[image]
[image]

**Building the AssetBundle**:
[image]

EricBoiseLGSVL, 2021-12-15:
Can you please check that the mesh renderer is referencing the mesh you have in the M2 folder and not in another directory?
Select the M2 mesh and right click.  Then select Prefab -> Select asset.  It should highlight the mesh in the M2 Vehicles folder.  Also make sure the textures in the materials reference the textures in the correct folder.

EricBoiseLGSVL, 2021-12-15:
If you are able, zip the vehicle and send to us here to look.

eslaaam3, 2021-12-16:
It is referencing the correct mesh in the directory
[image]

Building fails with the Jaguar as well! :(
[image]

However, the CubeTown example is built successfully

> If you are able, zip the vehicle and send to us here to look.

I'm not allowed to share the robot's CAD

EricBoiseLGSVL, 2021-12-16:
Darn, ok.  Something is not packaging the fbx correctly.  Last thing is to update the fbx exporter in unity packages to lastest.  It won;t work on ubuntu 18.04 btw

eslaaam3, 2021-12-19:
Unfortunately, this is also not working, I updated it to version 4.1.2 and I get the same error of gITF converter

EricBoiseLGSVL, 2021-12-20:
One last thing to try is to make sure your fbx is read/write enabled.  Select the fbx and check the option in the import menu