lgsvl/simulator #1919: Intersection and waypoint following

DestinyFrank, 2021-12-16:
We are working on designing a delivery robot that drives on the sidewalk. I am currently working on annotating intersections. Is there a way to make the vehicle drive along the waypoint lanes in the simulation? Also, is there a way to annotate the lanes so the vehicle can drive in either direction on the lane? 

[image]

EricBoiseLGSVL, 2021-12-16:
We have a VehicleWaypointFollowSMI.cs that you can replace the VehicleSMI.cs with and it will follow lanes like npcs.  No, we don't support driving in the opposite direction.  You will need to heavily edit the waypoint follow class.

DestinyFrank, 2021-12-17:
Where can I find the VehicleWaypointFollowSMI.cs script to use in the project?

EricBoiseLGSVL, 2021-12-17:
In the project under Assets/Scripts/Dynamics/Examples/VehicleWaypointFollowSMI.cs

DestinyFrank, 2021-12-17:
I found the script and added it to the vehicle. Do you need to plug in the map you are planning to use? Is there anything else I need to plug into the code? 

[image]
ipt how

EricBoiseLGSVL, 2021-12-17:
nope that looks good.  Just be sure to remove VehicleSMI.  You can only have one dynamics class on a vehicle.  Build the bundle and upload to webui as a different vehicle so you know which one has waypoint follow.  Then run a sim.  You may need to edit the class to get the desired behavior.

DestinyFrank, 2021-12-19:
It is still doesn't seem to work. How would I edit the class?

EricBoiseLGSVL, 2021-12-20:
What exactly doesn't work?

DestinyFrank, 2021-12-20:
The Vehicle doesn't move and stays in the spawn position when I run a simulation. Is there a way to edit the class to have six wheels? I have Visual Studios, and I removed the Vehicle SMI from the prefab.

EricBoiseLGSVL, 2021-12-21:
Did you remove the center wheels from the wheels data list and tick the steering for the front two?  Did you test on CubeTown or Borregas?

To support 6 wheels, which shouldn't be hard because the LaneFollowSMI moves the rigidbody, steering should just be for wheel angle.  Do you have any errors when you run?  What does the player.log say, after the above tests?

DestinyFrank, 2021-12-21:
Yes, I did turn off the middle wheels and change the steering. When I ran a test with the Borregas map, I still had the same issue. There does not appear to be any errors, but the vehicle still doesn't move.

EricBoiseLGSVL, 2021-12-21:
Hmm, we haven't seen this in our testing.  We used the Jag with the LaneFollow on Borregas.  Can you try with the Jag to see if it's an issue with how you created your EGO model? Are you using articulated bodies?

DestinyFrank, 2021-12-22:
I do not believe that I am using articulated bodies. Would I need to change the Configration to get the lane following to work?

EricBoiseLGSVL, 2021-12-22:
No that is good then.  We haven't tested VehicleLaneFollowSMI with articulated bodies.
What do you mean change the configuration?

Keyboard sensor will not work with LaneFollowing.  If it worked on your test then you are not using the correct vehicle.
Be very sure that your loaded vehicle is the lane follow smi. You can run in the editor and select the vehicle to check.

Vehicle lane follow will make the EGO behave like a NPC.  It will be moved by the rigidbody.  It will no longer use inputs and will randomly follow the closet lane.  It will despawn and respawn when it runs out of annotated lanes.

Be sure the spawn position is on a valid lane as well.  You can use any other sensor but control sensors.  If you want control with lane follow I suggest using an ad stack like apollo or autoware.  This is how the simulator is designed.