lgsvl/simulator #1922: Demo Requested

jackxia123, 2021-12-20:
i can continue the process, but it is a little stange, see below , is it normal?
Stopping lanefollowing_drive ... done
xiayu@xiayu-Lenovo-Legion-R7000P2021H:~/tmp/lanefollowing$ sudo docker-compose up drive_visual
Starting lanefollowing_drive_visual ... done
Attaching to lanefollowing_drive_visual
drive_visual_1 | 2021-12-16 08:24:55.070270: I tensorflow/core/platform/cpu_feature_guard.cc:140] Your CPU supports instructions that this TensorFlow binary was not compiled to use: AVX2 FMA
drive_visual_1 | 2021-12-16 08:24:55.179010: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:898] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero
drive_visual_1 | 2021-12-16 08:24:55.179335: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1356] Found device 0 with properties:
drive_visual_1 | name: NVIDIA GeForce RTX 3060 Laptop GPU major: 8 minor: 6 memoryClockRate(GHz): 1.425
drive_visual_1 | pciBusID: 0000:01:00.0
drive_visual_1 | totalMemory: 5.81GiB freeMemory: 3.04GiB
drive_visual_1 | 2021-12-16 08:24:55.179371: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1435] Adding visible gpu devices: 0
drive_visual_1 | 2021-12-16 08:28:38.787336: I tensorflow/core/common_runtime/gpu/gpu_device.cc:923] Device interconnect StreamExecutor with strength 1 edge matrix:
drive_visual_1 | 2021-12-16 08:28:38.787387: I tensorflow/core/common_runtime/gpu/gpu_device.cc:929] 0
drive_visual_1 | 2021-12-16 08:28:38.787394: I tensorflow/core/common_runtime/gpu/gpu_device.cc:942] 0: N
drive_visual_1 | 2021-12-16 08:28:38.788089: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1053] Created TensorFlow device (/job:localhost/replica:0/task:0/device:GPU:0 with 2920 MB memory) -> physical GPU (device: 0, name: NVIDIA GeForce RTX 3060 Laptop GPU, pci bus id: 0000:01:00.0, compute capability: 8.6)
drive_visual_1 | [INFO] [drive]: Starting Drive node...
drive_visual_1 | [INFO] [drive]: Camera topic: /simulator/sensor/camera/center/compressed
drive_visual_1 | [INFO] [drive]: Control topic: /lanefollowing/steering_cmd
drive_visual_1 | [INFO] [drive]: Loading model from /lanefollowing/ros2_ws/src/lane_following/model/model.h5
drive_visual_1 | [INFO] [drive]: Model loaded!
drive_visual_1 | [INFO] [drive]: Up and running...
drive_visual_1 | [INFO] [drive]: Waiting a camera image from ROS topic: /simulator/sensor/camera/center/compressed
drive_visual_1 | [INFO] [drive]: Waiting a camera image from ROS topic: /simulator/sensor/camera/center/compressed
drive_visual_1 | [INFO] [drive]: Waiting a camera image from ROS topic: /simulator/sensor/camera/center/compressed
drive_visual_1 | [INFO] [drive]: Waiting a camera image from ROS topic: /simulator/sensor/camera/center/compressed
drive_visual_1 | [INFO] [drive]: Waiting a camera image from ROS topic: /simulator/sensor/camera/center/compressed
drive_visual_1 | [INFO] [drive]: Waiting a camera image from ROS topic: /simulator/sensor/camera/center/compressed
drive_visual_1 | [INFO] [drive]: Waiting a camera image from ROS topic: /simulator/sensor/camera/center/compressed
drive_visual_1 | [INFO] [drive]: Waiting a camera image from ROS topic: /simulator/sensor/camera/center/compressed
drive_visual_1 | [INFO] [drive]: Waiting a camera image from ROS topic: /simulator/sensor/camera/center/compressed
drive_visual_1 | [INFO] [drive]: Waiting a camera image from ROS topic: /simulator/sensor/camera/center/compressed
drive_visual_1 | [INFO] [drive]: Waiting a camera image from ROS topic: /simulator/sensor/camera/center/compressed

EricBoiseLGSVL, 2021-12-20:
Sorry, what is your issue exactly?

jackxia123, 2021-12-27:
how to add new ego vehicle in svl , i can't see youtube video, because i am in China, could you send me other video download link?

EricBoiseLGSVL, 2022-01-04:
Sorry we haven't been able to upload the video.  Our docs have a step by step guide the follows the video, please use that.  I also recommend cloning the Jag repo as an example.  By duplicating how the Jag is setup, you should be fine.  Just remember to relink public vars in the inspector, they might link to the jag when you duplicate objects.

jim18324, 2022-01-13:
There is video on Bilibili
https://www.bilibili.com/video/BV1bD4y1S75N?spm_id_from=333.999.0.0

The Jag repo:
https://github.com/lgsvl/Jaguar2015XE