lgsvl/simulator #1923: fail to install SVL

frankie-wei, 2021-12-20:
I use GPU resource in alibaba cloud. I install Ubuntu 20.04（64-bit）first then install windows 10 as double system OS.

the GPU model as：GN6I 4 core 15 GB memory 1 NVIDIA T4

it encounter below error message

[image]

try to install gpu drive firtst then install CUDA and cudnn and LgSVL but still fail。

[image]

I also run nvidia-smi to check gpu status.

please kindly advise how to resolve the failure problem

[image]

EricBoiseLGSVL, 2021-12-20:
Why are you installing windows and Linux?  To just run the simulator executable, you only need the windows or the linux application.  I recommend you use our 2021.3 release build and run on the OS for the build you have.  After that, the GPU may have issues running Unity but should be fine unless it doesn't support the vulkan drivers.

frankie-wei, 2021-12-21:
> Why are you installing windows and Linux? To just run the simulator executable, you only need the windows or the linux application. I recommend you use our 2021.3 release build and run on the OS for [...]

Eric, thanks for your prompt support. May I know if it could install 2021.3 release for linux on Ubuntu 20.04（64-bit）?

EricBoiseLGSVL, 2021-12-21:
Yes, but I am confused on the install question.  Are you installing Unity Editor or just running the 2021.3 release binary?

frankie-wei, 2021-12-22:
Eric, would you please help to advice the correct steps to install SVLsimulator

EricBoiseLGSVL, 2021-12-22:
Sure but what do you mean install?  SVLSimulator can be a binary that you run on your machine or you can install the Unity Editor and clone svlsimulator repositories if you want to build custom svlsimulator binaries or custom assets.
Keep in mind we do not support running the Unity editor in the cloud, is this what you are trying to do?