lgsvl/simulator #1925: Visual Studios New Map

DestinyFrank, 2021-12-20:
I am learning to create a simulation using the Visual Editor. I want to use a new map from my library that I created and uploaded to the website. Is there a way to add the new to the Maps in the Editor? 

[image]

EricBoiseLGSVL, 2021-12-21:
Sure, VSE will have access if you add to your library.  I always run a simulation with the map locally from the webui and then VSE has the map cached.

DestinyFrank, 2021-12-21:
I have added the new map to my Libary and run the map in simulation. It was still not showing in the VSE options.

EricBoiseLGSVL, 2021-12-21:
No, unless you set data folder to a different path.  What data path is the exe and editor on?

DestinyFrank, 2021-12-21:
Do you mean it needs to be in the exact file location? I apologize; I am unsure about the data path.

EricBoiseLGSVL, 2021-12-21:
No worries.  So we have a config.yml file we use to specify a data path if users don't want to use the default.  If you don't then the data path is set to application.presistentdatapath from Unity apis.  So it would be like so.
[image]

This is were you cache the assets you pull from the webui.

My concern is that you are running the exe and editor to different data paths so VSE will not have it cached.  If you haven't altered a config then the exe and unity editor should be on the same path.

The next thing to check if the map is public.  Try with a test map to be sure or an existing map we provide.  Last would be to clear cache of maps with the cache control menu in the main menu.  Clear all local in case you have some name conflict and try to run the sim again, then vse

DestinyFrank, 2021-12-22:
Here is my folder with the simulator is located. I am unsure if this is correct or not. My map is private, but my other private maps show in the VSE already. 

Also, would I clear the cache in the VSE or the website? 

[image]

EricBoiseLGSVL, 2021-12-22:
No this is not correct.
Simulator repo should be separate from the built binary.
Do not build the binary inside the repository source.
SVL Source:
[image]

Simulator asset source in simulator repo:
[image]

Simulator appdata folder with cached files and database:
[image]

Simulator Build:
[image]

Run the unity editor simulator (after you remove build files) and start with LoaderScene.scene in Assets/Scenes/LoaderScene
Only run binary once you are done developing and want to run end to end tests with AD stack to have better performance.

EricBoiseLGSVL, 2021-12-22:
When I clear cache after large changes in source I have a few steps.
1- I remove all assets in WebUI that relate to changes. Start fresh to make sure all cached assets are latest
2 - I delete cached asset in simulator, run loaderscene, go to drop down menu top right and select cache control, delete cached assets.  You can also just delete the entire appdata folder and relink the cluster.
3 - I build new assets and upload to webui.
4 - Create new simulations that use the new assets in webui.
Then unity editor will have all new cached assets.