lgsvl/simulator #1926: Difference between sensor timestamp info and my computer's clock

gizem-buss, 2021-12-21:
Hi,
I have a problem with sensors with header information. The timestamp information coming from the sensors does not match the clock of my computer. From which source is the timestamp information on the header of the sensors taken?

Thank you.

EricBoiseLGSVL, 2021-12-21:
Simulator time.  This way AD stacks don't have lags in time (Apollo and Autoware expect to run real time in a vehicle).  This is why we have the Clock Sensor.  Please see the documentation [here](https://www.svlsimulator.com/docs/simulation-content/sensors-list/#clock), [here](https://www.svlsimulator.com/docs/tutorials/robotics-pub-sub/#clock) and [here](https://www.svlsimulator.com/docs/system-under-test/apollo-master-instructions/#setting-clock-mode).

gizem-buss, 2021-12-22:
Thank you.,Can I use real time instead of simulator time? Is there a feature that enables this?

closed 2021-12-22 by EricBoiseLGSVL