lgsvl/simulator #1935: Unable to build new ego vehicle (gITF converter failed with exit code: 105)

FiIipe, 2022-01-04:
Hello, 

I've been trying to create a new ego vehicle to use with LGSVL. 
The vehicle that I'm trying to import is the Kia Niro and this is the .fbx file. 

[Kia.zip](https://github.com/lgsvl/simulator/files/7807711/Kia.zip)

The version of unity that I'm using is the 2020.3.3f1, the version of the simulator is the 2021.3 and the version of the Jaguar is also the 2021.3.

I've followed this tutorial and the video: https://www.svlsimulator.com/docs/simulation-content/add-new-ego-vehicle/

When I press build, I get the following errors: 

[image]

Any help is very much appreciated. 

Cheers, 
Filipe

EricBoiseLGSVL, 2022-01-04:
What OS and version are you running?

FiIipe, 2022-01-04:
@EricBoiseLGSVL thank you for your quick response. I'm running on windows 10

EricBoiseLGSVL, 2022-01-04:
hmm, we have not seen this issue internally but I have seen a few users with this issue. Let me look at the model

EricBoiseLGSVL, 2022-01-04:
Did you make the fbx read write enabled?
[image]

FiIipe, 2022-01-04:
No, I did not change that field. Should I try again with read/write enabled?

EricBoiseLGSVL, 2022-01-04:
yes please try that, then I have one more thing.  Be sure you export the material in the fbx importer.  I am building as well test.

FiIipe, 2022-01-04:
Ok, I will try it. 
I'm not sure what you mean by this: "Be sure you export the material in the fbx importer". Can you explain, please?

EricBoiseLGSVL, 2022-01-04:
I bet this is the issue.  I have successfully built the bundle locally.
The gltf exporter needs access to the materials to make changes to support the glb file.  The folders need to look like this:
[image]

The export option is here in the fbx file like so:
[image]

This will extract the material data to the folder you make in Models.  Then, the exporter has access to the materials.

FiIipe, 2022-01-05:
@EricBoiseLGSVL The error still persists. I don't think the problem is the prefab or the fbx model. I tried to build the Jaguar model and I get exactly the same error. Can the problem be something wrong with the my unity installation or the simulator build? 

[image]

If you managed to build the model, can you make it available on the LGSVL vehicle store? 

Cheers, 
Filipe

EricBoiseLGSVL, 2022-01-05:
Can you show me the full path to the assetbundles folder?  The path seems strange.  Show the path to the assets you are building and the asset path to the AssetBundles folder like so:
[image]

and
[image]

Sorry since I don't own the rights to the model I can't publish it.  You should also edit your previous post with the fbx

FiIipe, 2022-01-05:
@EricBoiseLGSVL ok, I just removed the  .fbx model of the kia.

The path to the AssetBundles folder is the following: 

[image]

Inside AssetBundles -> vehicles:

[image]

The path to the fbx file: 

[image]

Is there anyway that you can send me the file instead of publishing it?

Cheers, Filipe

FiIipe, 2022-01-05:
@EricBoiseLGSVL, I was able to build it, finally!

I was trying to build the vehicle in the following directory: `C:\Users\Filipe Cerveira\Desktop\simulator`. 
I moved the entire folder from that directory to the root of disk C, `C:\simulator`, and I tried to build it again. Everything worked smoothly, no problems. It seems like the space between Filipe Cerveira was messing up the build process for some unknown reason. I will try to build it again in a directory with space to make sure that the problem was the space.

EricBoiseLGSVL, 2022-01-06:
lol excellent. yes the space was probably the issue.  Glad you found a solution and thanks for posting what worked for others.

closed 2022-01-06 by EricBoiseLGSVL