lgsvl/simulator #1940: Simulator error "Bad Gateway"

JordanCrawfordx, 2022-01-06:
Hello,
 
I'm trying to run a simulation but there seems to be an error with the gateway. Every time I run, the error message shows. Here's the error:

<html> <head><title>502 Bad Gateway</title></head> <body> <center><h1>502 Bad Gateway</h1></center> <hr><center>nginx/1.20.1</center> </body> </html>      

Also, I'm unable to view the image of my simulation as shown here: 

[image]

@EricBoiseLGSVL If you could help, it'd be greatly appreciated! :)

EricBoiseLGSVL, 2022-01-06:
Looks like an issue with your map, you should have a view of the scene instead of the generic image.  Please post the playerlog and screen shots of the map hierarchy panel in Unity.  Also, any errors you get when building it.
FYI windows 11 users have been reporting issues with simulator.  It is not officially supported.

EricBoiseLGSVL, 2022-01-06:
This also may be a backend issue, see #1938 and #1937.  We are looking into it now, thanks for posting

EricBoiseLGSVL, 2022-01-06:
Fixed, thanks for posting.  reopen the issue if you still have issues.  Seems a node crashed without a warning.

closed 2022-01-06 by EricBoiseLGSVL

JordanCrawfordx, 2022-01-07:
It works! :D Thanks for the fix.