lgsvl/simulator #1942: npm install error in Ubuntu 20.04

Carlzhangk, 2022-01-11:
I followed the tutorial at https://gitlab.com/ApexAI/autowareclass2020/-/blob/master/lectures/11_LGSVL_Simulator/lgsvl-simulation.md#ui and Installed nodejs. The command npm install has the following errors. Any suggestion will be greatly appreciated.
My OS is Ubuntu 20.04.

TypeError: Cannot read property 'replace' of undefined
    at [eval]:1:43
    at Script.runInThisContext (vm.js:122:20)
    at Object.runInThisContext (vm.js:329:38)
    at Object.<anonymous> ([eval]-wrapper:6:22)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
    at evalScript (internal/bootstrap/node.js:590:27)
    at startup (internal/bootstrap/node.js:265:9)
    at bootstrapNodeJSCore (internal/bootstrap/node.js:623:3)
gyp: Call to 'node -e "console.log(process.env.AMENT_PREFIX_PATH.replace(/:/g, '/include/ ') + '/include/')"' returned exit status 1 while in binding.gyp. while trying to load binding.gyp
gyp ERR! configure error 
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/usr/share/nodejs/node-gyp/lib/configure.js:355:16)
gyp ERR! stack     at ChildProcess.emit (events.js:198:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:248:12)
gyp ERR! System Linux 5.11.0-46-generic
gyp ERR! command "/usr/bin/node" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /home/kai/adehome/ros2-web-bridge/node_modules/rclnodejs
gyp ERR! node -v v10.19.0
gyp ERR! node-gyp -v v6.1.0
gyp ERR! not ok 
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! rclnodejs@0.10.3 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the rclnodejs@0.10.3 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/kai/.npm/_logs/2022-01-11T02_29_16_188Z-debug.log

EricBoiseLGSVL, 2022-01-13:
If you are using the latest release npm is not needed.  The version of simulator that uses npm is not supported.

Carlzhangk, 2022-01-16:
@EricBoiseLGSVL Thank you for answering the issue. What is your advice to run "node bin/rosbridge.js"? When I run the following node, it asked the module 'rclnodejs' but based the suggestion at https://www.npmjs.com/package/rclnodejs/v/0.21.0, I need npm to install rclnodejs. 

kai@ade:~/ros2-web-bridge ((0.2.7))$ node bin/rosbridge.js
internal/modules/cjs/loader.js:638
    throw err;
    ^

Error: Cannot find module 'rclnodejs'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:636:15)
    at Function.Module._load (internal/modules/cjs/loader.js:562:25)
    at Module.require (internal/modules/cjs/loader.js:692:17)
    at require (internal/modules/cjs/helpers.js:25:18)
    at Object.<anonymous> (/home/kai/ros2-web-bridge/index.js:17:19)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)