lgsvl/simulator #1943: Stuck at starting screen

AlperenKosem, 2022-01-11:
After starting simulator, stuck at starting scree44444n. Offline button and link to cloud button are not working. Nvidia drivers are updated and there is no mesa vulkan drivers. 

**Output:**
```
$ ./simulator          
Set current directory to /home/alperen/simulation/simulators/svlsimulator-linux64-2021.3
Found path: /home/alperen/simulation/simulators/svlsimulator-linux64-2021.3/simulator
```

**Specifications :**
- Ubuntu 20.04
- 11th Gen Intel® Core™ i7-11800H @ 2.30GHz × 16 
- NVIDIA GeForce RTX 3060 Laptop GPU
- Driver Version: 495.46 

Player Log:
---------------------------------------------------------

Mono path[0] = '/home/alperen/simulation/simulators/svlsimulator-linux64-2021.3/simulator_Data/Managed'
Mono config path = '/home/alperen/simulation/simulators/svlsimulator-linux64-2021.3/simulator_Data/MonoBleedingEdge/etc'
Preloaded 'UnityFbxSdkNative.so'
Preloaded 'liblaszip.so'
Preloaded 'libpng.so'
Preloaded 'libsqlite3.so'
Preloaded 'libturbojpeg.so'
Display 0 '0': 1920x1080 (primary device).
Display 1 'VX3276-QHD 32"': 2560x1440 (secondary device).
Loading player data from /home/alperen/simulation/simulators/svlsimulator-linux64-2021.3/simulator_Data/data.unity3d
Desktop is 2560 x 1440 @ 60 Hz
[Vulkan init] extensions: count=18
[Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_display_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=1
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_surface_protected_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_wayland_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
Vulkan detection: 2
Initialize engine version: 2020.3.3f1 (76626098c1c4)
[Subsystems] Discovering subsystems at path /home/alperen/simulation/simulators/svlsimulator-linux64-2021.3/simulator_Data/UnitySubsystems
GfxDevice: creating device client; threaded=1
[Vulkan init] extensions: count=18
[Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_display_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=1
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_surface_protected_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_wayland_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[... 170 lines cut ...]

ArgumentOutOfRangeException: Index was out of range. Must be non-negative and less than the size of the collection.
Parameter name: index
  at System.ThrowHelper.ThrowArgumentOutOfRangeException (System.ExceptionArgument argument, System.ExceptionResource resource) [0x00029] in <eae584ce26bc40229c1b1aa476bfa589>:0 
  at System.ThrowHelper.ThrowArgumentOutOfRangeException () [0x00000] in <eae584ce26bc40229c1b1aa476bfa589>:0 
  at CloudAPI.GetInfo () [0x001c4] in <33a3e48d8c314f54845dfb34c374a12a>:0 
  at Simulator.Web.ConnectionUI.Awake () [0x00019] in <33a3e48d8c314f54845dfb34c374a12a>:0 

ArgumentNullException: Value cannot be null.
Parameter name: configuration
  at PetaPoco.Database..ctor (PetaPoco.IDatabaseBuildConfiguration configuration) [0x00016] in <d3306f055f6c490aada69d8080999d36>:0 
  at PetaPoco.DatabaseConfigurationExtensions.Create (PetaPoco.IDatabaseBuildConfiguration source) [0x00001] in <d3306f055f6c490aada69d8080999d36>:0 
  at Simulator.Database.DatabaseManager.Open () [0x00000] in <33a3e48d8c314f54845dfb34c374a12a>:0 
  at Simulator.Database.Services.ClientSettingsService.GetOrMake () [0x00000] in <33a3e48d8c314f54845dfb34c374a12a>:0 
  at ConnectionManager.Start () [0x00030] in <33a3e48d8c314f54845dfb34c374a12a>:0 

Setting up 8 worker threads for Enlighten.
  Thread -> id: 7fa6f7fff700 -> priority: 1 
  Thread -> id: 7fa6f77fe700 -> priority: 1 
  Thread -> id: 7fa6f6ffd700 -> priority: 1 
  Thread -> id: 7fa6f67fc700 -> priority: 1 
  Thread -> id: 7fa6f5ffb700 -> priority: 1 
  Thread -> id: 7fa6f57fa700 -> priority: 1 
  Thread -> id: 7fa6f4ff9700 -> priority: 1 
  Thread -> id: 7fa6f47f8700 -> priority: 1

EricBoiseLGSVL, 2022-01-13:
Have you opened the svlsimulator web ui and signed in before trying to link the simulator binary on the machine?

AlperenKosem, 2022-01-14:
> Have you opened the svlsimulator web ui and signed in before trying to link the simulator binary on the machine?

yes, i did.

EricBoiseLGSVL, 2022-01-14:
Please screen shot the main menu.  I am not sure what exactly is the issue.  Are you upgrading from an older version?  Is this in the binary or the editor?

AlperenKosem, 2022-01-17:
I am not upgrading new version. I can not link simulator to cluster.
1)
[image]

2) All of buttons are not working, not giving any response
[image]

EricBoiseLGSVL, 2022-01-17:
Looks like you are offline.  Are you running the app as admin?  Do you have any software preventing internet access?

EricBoiseLGSVL, 2022-01-17:
I would also delete the LGElectronics folder in AppData. Maybe the database is corrupted.

EricBoiseLGSVL, 2022-01-17:
Last thing, what web browser are you using?

githjk, 2022-06-05:
> After starting simulator, Offline button and link to cloud button are not working. Nvidia drivers are updated and there is no mesa vulkan drivers.
> 
> **Output:**
> 
> ```
> $ ./simulator          
> Set current directory to /home/alperen/simulation/simulators/svlsimulator-linux64-2021.3
> Found path: /home/alperen/simulation/simulators/svlsimulator-linux64-2021.3/simulator
> ```
> 
> **Specifications :**
> 
> * Ubuntu 20.04
> * 11th Gen Intel® Core™ i7-11800H @ 2.30GHz × 16
> * NVIDIA GeForce RTX 3060 Laptop GPU
> * Driver Version: 495.46
> 
> ## Player Log:
> Mono path[0] = '/home/alperen/simulation/simulators/svlsimulator-linux64-2021.3/simulator_Data/Managed' Mono config path = '/home/alperen/simulation/simulators/svlsimulator-linux64-2021.3/simulator_Data/MonoBleedingEdge/etc' Preloaded 'UnityFbxSdkNative.so' Preloaded 'liblaszip.so' Preloaded 'libpng.so' Preloaded 'libsqlite3.so' Preloaded 'libturbojpeg.so' Display 0 '0': 1920x1080 (primary device). Display 1 'VX3276-QHD 32"': 2560x1440 (secondary device). Loading player data from /home/alperen/simulation/simulators/svlsimulator-linux64-2021.3/simulator_Data/data.unity3d Desktop is 2560 x 1440 @ 60 Hz [Vulkan init] extensions: count=18 [Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0 [Vulkan init] extensions: name=VK_KHR_display, enabled=1 [Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0 [Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0 [Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0 [Vulkan init] extensions: name=VK_KHR_get_display_properties2, enabled=0 [Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=1 [Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0 [Vulkan init] extensions: name=VK_KHR_surface, enabled=1 [Vulkan init] extensions: name=VK_KHR_surface_protected_capabilities, enabled=0 [Vulkan init] extensions: name=VK_KHR_wayland_surface, enabled=1 [Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0 [Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1 [Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0 [Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0 [Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0 [Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0 [Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0 Vulkan detection: 2 Initialize engine version: 2020.3.3f1 (76626098c1c4) [Subsystems] Discovering subsystems at path /home/alperen/simulation/simulators/svlsimulator-linux64-2021.3/simulator_Data/UnitySubsystems GfxDevice: creating device client; threaded=1 [Vulkan init] extensions: count=18 [Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0 [Vulkan init] extensions: name=VK_KHR_display, enabled=1 [Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0 [Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0 [Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0 [Vulkan init] extensions: name=VK_KHR_get_display_properties2, enabled=0 [Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=1 [Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0 [Vulkan init] extensions: name=VK_KHR_surface, enabled=1 [Vulkan init] extensions: name=VK_KHR_surface_protected_capabilities, enabled=0 [Vulkan init] extensions: name=VK_KHR_wayland_surface, enabled=1 [Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0 [Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1 [Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0 [Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0 [Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0 [Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0 [Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0 [Vulkan init] SelectPhysicalDevice requestedDeviceIndex=-1 xrDevice=(nil) [Vulkan init] Physical Device 0x2377ea0 [0]: "NVIDIA GeForce RTX 3060 Laptop GPU" deviceType=2 vendorID=10de deviceID=2520 [Vulkan init] Selected physical device 0x2377ea0 [Vulkan init] Graphics queue count=1 [Vulkan init] extensions: count=145 [Vulkan init] extensions: name=VK_KHR_16bit_storage, enabled=0 [Vulkan init] extensions: name=VK_KHR_8bit_storage, enabled=0 [Vulkan init] extensions: name=VK_KHR_acceleration_structure, enabled=0 [Vulkan init] extensions: name=VK_KHR_bind_memory2, enabled=0 [Vulkan init] extensions: name=VK_KHR_buffer_device_address, enabled=0 [Vulkan init] extensions: name=VK_KHR_copy_commands2, enabled=0 [Vulkan init] extensions: name=VK_KHR_create_renderpass2, enabled=1 [Vulkan init] extensions: name=VK_KHR_dedicated_allocation, enabled=1 [Vulkan init] extensions: name=VK_KHR_deferred_host_operations, enabled=0 [Vulkan init] extensions: name=VK_KHR_depth_stencil_resolve, enabled=1 [Vulkan init] extensions: name=VK_KHR_descriptor_update_template, enabled=1 [Vulkan init] extensions: name=VK_KHR_device_group, enabled=0 [Vulkan init] extensions: name=VK_KHR_draw_indirect_count, enabled=0 [Vulkan init] extensions: name=VK_KHR_driver_properties, enabled=0 [Vulkan init] extensions: name=VK_KHR_external_fence, enabled=0 [Vulkan init] extensions: name=VK_KHR_external_fence_fd, enabled=0 [Vulkan init] extensions: name=VK_KHR_external_memory, enabled=0 [Vulkan init] extensi [... cut ...] tionManager.Start () [0x00030] in <33a3e48d8c314f54845dfb34c374a12a>:0
> 
> Setting up 8 worker threads for Enlighten. Thread -> id: 7fa6f7fff700 -> priority: 1 Thread -> id: 7fa6f77fe700 -> priority: 1 Thread -> id: 7fa6f6ffd700 -> priority: 1 Thread -> id: 7fa6f67fc700 -> priority: 1 Thread -> id: 7fa6f5ffb700 -> priority: 1 Thread -> id: 7fa6f57fa700 -> priority: 1 Thread -> id: 7fa6f4ff9700 -> priority: 1 Thread -> id: 7fa6f47f8700 -> priority: 1

I have the same problem, have you solve it ?

githjk, 2022-06-05:
> > Have you opened the svlsimulator web ui and signed in before trying to link the simulator binary on the machine?
> 
> yes, i did.

I have the same problem, have you solved it?

AlperenKosem, 2022-06-05:
> > > Have you opened the svlsimulator web ui and signed in before trying to link the simulator binary on the machine?
> > 
> > 
> > yes, i did.
> 
> I have the same problem, have you solved it?

i solved by instrunctions of third answer in this forum:

https://forum.unity.com/threads/unable-to-login-unity-hub-on-linux.1176971/

DOMINICQQQ, 2022-07-10:
> > > > 在尝试链接机器上的模拟器二进制文件之前，您是否打开了 svlsimulator web ui 并登录？
> > > 
> > > 
> > > 是的，我做到了。
> > 
> > 
> > 我也有同样的问题，你解决了吗？
> 
> 我通过本论坛中第三个答案的说明解决了：
> 
> https://forum.unity.com/threads/unable-to-login-unity-hub-on-linux.1176971/

Hi! can you please tell me the specific solution? i have the same question

avarebiri, 2023-01-23:
ı have the same problem did you solve it ?