lgsvl/simulator #1945: how to add a new vehicle model

jackxia123, 2022-01-12:
what is jag, docs has less steps when i try to add a new vehicle model into lgsvl, video is strightforward, and our company is the biggest automative industry in China, if we use your tool into production, your tool will be very famous

EricBoiseLGSVL, 2022-01-13:
The jaguar is a public repo, [here](https://github.com/lgsvl/Jaguar2015XE), that we provide to help follow our documentation on how to create a new vehicle.  Glad you like the software, we are happy that it may help you and your team.

jackxia123, 2022-01-18:
cdsdc

jackxia123, 2022-01-18:
very cool, i just git clone this project and build in Unity 3D, and try to upload it to my Library/Vehicle, but it threw "Unable to read archive entries:Bad archive"

EricBoiseLGSVL, 2022-01-18:
Be sure you have pulled with git lfs installed and installed both windows and linux mono build support in Unity Hub