lgsvl/simulator #1950: How to communicate with custom client using custom brigde plugin?

umdreamer, 2022-01-14:
I want to communicate simulator with my custom client (e.g. a client can send message to control ego vehcile and receive ego vehcile state messages) using custom bridge plugin. I found in #930 https://github.com/lgsvl/simulator/issues/930, they can use websocket to achieve this.

In the documentation, I only found how to write a custom bridge plugin in Simulator. But how I can communicate with this bridge in my custom client. Could give me some suggestion how to do? What should I do in the custom client? Thanks.

EricBoiseLGSVL, 2022-01-14:
What do you mean custom client?  A custom simulator build?  What are you trying to accomplish?

umdreamer, 2022-01-15:
Our custom client is another program which is used to send and receive vehicle data. That is, I want to use this program to control the vehicle in the Simulator (which is built from source based on 2021.3) throgh custom bridge.

umdreamer, 2022-01-15:
> What do you mean custom client? A custom simulator build? What are you trying to accomplish?

Our custom client is another program which is used to send and receive vehicle data. That is, I want to use this program to control the vehicle in the Simulator (which is built from source based on 2021.3) throgh custom bridge.

umdreamer, 2022-01-15:
Hi @EricBoiseLGSVL ,

To explain more about my question. I view the Simulator as the server. And I want to send and receive data to/from the Simulator using a client written by Python API. In Python API, I can connect the bridge by `ego.connect_bridge('127.0.0.1', 9090)`. And I can make sure the bridge is connected by checking the status of `ego.bridge_connected`.

Then after that, how can I get data from the ego sensors? For example, how can I receive images from ColorCameraSensor? and How can I receive data from CanBusSensor?

Thank you.

umdreamer, 2022-01-17:
The documentation is https://www.svlsimulator.com/docs/tutorials/create-ros2-ad-stack/. It is helpful:)
> After read again the documentation, I finally understood that the communication between Simualtor and client is through ROS2 messages. After diving into a bit of the ROS, I worked out my problem. Now, I can send and receive messages to and from Simulator.