lgsvl/simulator #1956: How to control the ego with VehicleControlData?

umdreamer, 2022-01-18:
Hi, I find the examples in the documation to control the ego vehicle using ROS2 message.  I add a Vehicle Control Sensor to the vehicle. The message type is VehicleControlData. which can be displayed by `cat cat lgsvl_msgs/msg/VehicleControlData.msg`:
```
std_msgs/Header header

float32 acceleration_pct  # 0 to 1
float32 braking_pct  # 0 to 1
float32 target_wheel_angle  # radians
float32 target_wheel_angular_rate  # radians / second
uint8 target_gear

uint8 GEAR_NEUTRAL = 0
uint8 GEAR_DRIVE = 1
uint8 GEAR_REVERSE = 2
uint8 GEAR_PARKING = 3
uint8 GEAR_LOW = 4
```
From here, I set target_wheel_angular_rate = 0.5, but it seems to be no effect on the vehicle. How should I do to  add/reduce speed, trun around of the vehicle? Thanks.

EricBoiseLGSVL, 2022-01-19:
Have you looked at the LGSVLControlSensor yet? [Here](https://www.svlsimulator.com/docs/simulation-content/sensors-list/#vehicle-control).

umdreamer, 2022-01-24:
> Have you looked at the LGSVLControlSensor yet? [Here](https://www.svlsimulator.com/docs/simulation-content/sensors-list/#vehicle-control).

Yes, I found it. I finally make it. Thank you.

closed 2022-01-24 by EricBoiseLGSVL