lgsvl/simulator #1957: unable to read archive entries:Bad archive

jackxia123, 2022-01-18:
i just used Unity 3d to build my new Vehicle dynamic , and build successfully, and try to add Vehicles in Library from Assets/External/Vehicles/myvechicle, it threw unable to read archive entries:Bad archive error, i just tried to extract my vehicle archive files, it threw bad archive errors, do you know the root cause  and how to solve it

JeyP4, 2022-01-18:
[image]

Make sure you have only these two modules.
I checked on windows OS.

EricBoiseLGSVL, 2022-01-19:
you don't need IL2CPP for windows.
What does the bundle look like after building?  Can you post here or at least a screen shot of the contents?