lgsvl/simulator #1959: Vehicle physics pro (VPP) vehicle import?

JeyP4, 2022-01-18:
Hello

I have been through the below posts
[LGSVL-VPP (Vehicle Physics Pro) Integration #515
](https://github.com/lgsvl/simulator/issues/515)&
[Post#1113
](https://github.com/lgsvl/simulator/issues/1113#issuecomment-740113330)
But they couldn't lead to a deterministic approach.

For good autonomous driving simulation, detailed vehicle dynamics is better to have.
Those posts are also old. I was wondering if there is any more-updated approach is available to import VPP vehicles in LGSVL.

Could anyone suggest something regarding it?

EricBoiseLGSVL, 2022-01-19:
Are you looking for the center of mass?

JeyP4, 2022-01-19:
> Are you looking for the center of mass?

No, I am looking for rotational Inertia which is also known as moment of inertia kg-m2.

This is missing in Jaguar prefab. This is a critical parameter in real vehicle dynamics.

EricBoiseLGSVL, 2022-01-19:
So [this](https://docs.unity3d.com/ScriptReference/Rigidbody-inertiaTensor.html) and/or [this](https://docs.unity3d.com/ScriptReference/Rigidbody-inertiaTensorRotation.html)?

JeyP4, 2022-01-19:
> So [this](https://docs.unity3d.com/ScriptReference/Rigidbody-inertiaTensor.html) and/or [this](https://docs.unity3d.com/ScriptReference/Rigidbody-inertiaTensorRotation.html)?

Thanks for replying.
Yes, I am aware of rigid body tensor. But they are automatically calculated by the mesh collider. And mesh collider assumes uniform weight distribution (which is an overestimation). That's why in the single-track vehicle model,  a reasonable `J` is used.

[image]

For example, in Vehicle physics pro correct J value is 2200 kg-m2. Rigid body collider calculates ~3100 kg-m2.
[image]

EricBoiseLGSVL, 2022-01-20:
Ah, I see. I wasn't aware it allowed you to set this. Yes, VPP is great.  We have integrated VPP but we can't release because of license issues. Some other users have as well.

We implemented custom dynamics and VPP can be added to the binary.  What is the issue that you are having integrating exactly?

JeyP4, 2022-01-20:
> What is the issue that you are having integrating exactly?

I have converted VPP material to HDRP. So now it is ready to be seen in LGSVL scene in unity.
Then I dragged Jaguar and VPP vehicle in same scene to compare them.

EricBoiseLGSVL, 2022-01-21:
Like the post you linked earlier, you need to create a new dynamicsSMI class that works with VPP.

lemketron, 2022-01-21:
@JeyP4 
> Basically, I am confused which are the next steps for me.

You can check out my VehicleVPP.cs file in a VPP fork of SVL that includes the (free) VPP community DLL here:
https://github.com/lemketron/simulator/tree/WIP-AddVPP-on-2021.3

This work was used by some of the [Indy Autonomous Challenge](https://www.indyautonomouschallenge.com/) teams for simulation prior to their autonomous racing events in Indianapolis last October and at CES in Las Vegas in January, 2022.

Hope this helps.

It would be great to have an open source vehicle with the VPP changes in it, but unfortunately we don't have anything available that we can share publicly share.

EricBoiseLGSVL, 2022-01-21:
Thanks @lemketron

JeyP4, 2022-01-22:
Thanks @lemketron

I tried using the scripts you suggested in the following way.
[image]

So some scripts from VPP default, some from your repo, and some from Jaguar. Is it the correct way?
But in simulation, the vehicle is sinking inside the ground.
[image]

I think it stops when body 'Collider' hits the ground.
Could you suggest something?

Thanks ! 🙏😀

EricBoiseLGSVL, 2022-01-24:
You need to change the code to support VPP wheel colliders, not Unity wheel colliders.  Make sure it is on the default physics layer.  Also, fix all null errors.

lemketron, 2022-01-24:
> You need to change the code to support VPP wheel colliders, not Unity wheel colliders. Make sure it is on the default physics layer. Also, fix all null errors.

This might help: https://vehiclephysics.com/components/wheel-collider/