lgsvl/simulator #1965: LGSVL setup in ubuntu in headless mode

tecena, 2022-01-20:
Hi,
I am trying to setup lgsvl simulator in an ubuntu machine. I do not have any screens for this machine the only way to access the machine is through terminal. Only using the commands. I can not see UI/ any other screens of the machine. 
So now I want to set up lgsvl simulator in this machine in an headless mode. .so the only way to access the simulator is through terminal. I can not access the UI of the simulator anymore.
So is it possible to set up simulator in this way in an headless mode with out any access to UI of the simulator and run it .Please suggest.
Thank you

lemketron, 2022-01-20:
The best and easiest suggestion I know of is to look into Nice-DCV:
https://docs.aws.amazon.com/dcv/latest/adminguide/what-is-dcv.html

tecena, 2022-01-21:
@lemketron 
Thank you for response. I think the above question did not give a clear meaning. I will try to explain the other way. We are using GPU enabled azure ubuntu machine which does not have RDP or any GUI. Now we want to install SVL Simulator in this machine. How can we install SVL Simulator   without GUI. We will be using wise APIs to link to cloud ,to start a simulation and to run it. We do not need SVL GUI. Is there any way we can setup simulator without GUI.
Please help

lemketron, 2022-01-21:
We never supported running headless so I'm not sure how to do the initial connection to WISE from the command line. You might be able to do it on a machine that does have a GUI, and then pack up and move the simulator data folder and install it on the headless remote machine...?

lemketron, 2022-01-21:
NICE-DCV is a commercial product (if not running on AWS) that supports full RDP-like remote access to a headless remote machine like you mentioned. I don't know if NICE-DCV works on Azure and you'd likely need a license to use it there if it did, but it might be worth it if you can't figure out another way to get those machines configured. But check out what I wrote previously (moving the data folder) as that may work for you.