lgsvl/simulator #1968: how to send apollo steering and long acc velocity to lgsvl(i mean interface alignment)

jackxia123, 2022-01-24:
Apollo + lgsvl can run successfully, i can;t find any document to introduce how to link apollo and lgsvl, i mean how to send apollo steering and long acc velocity to lgsvl , i want to send my control command to lgsvl and can run

EricBoiseLGSVL, 2022-01-24:
Apollo control sensor with cyber bridge.

lemketron, 2022-01-24:
See SVL Apollo documentation here: https://www.svlsimulator.com/docs/system-under-test/apollo-master-instructions/

jackxia123, 2022-02-15:
currently i can run lgsvl+apollo6.0 successfully, but i want to change apollo control module with my control module, i want to know how to change ,
it will cover interface alignment(lgsvl+my control module)

lemketron, 2022-02-15:
If your question is about how to change the Apollo control module you might want to ask it on the Apollo project (after reviewing the Apollo project documentation).
https://github.com/ApolloAuto/apollo/issues