lgsvl/simulator #1969: [SVL continue] new trail organization and repo for the SVL simulator to live on :)

ntutangyun, 2022-01-25:
Hi SVL team.

Thank you so much for all the contributions of the SVL simulator. It meant a lot to the community as well as to my research work.  

Since the current repo is not being maintained and there may be many developers like me pending new features etc. 
I hope you don't mind me copying most of the simulator related source code to this organization: 

https://github.com/SVL-simulator

the new simulator repository is https://github.com/SVL-simulator/simulator.

Here are some of my plans for the new repo: 

1. I've disabled all the git large file storage for the new simulator repository. all the files will be managed by git. As the large file storage seems to be non-free...
2. The new repo will be driven by the community (including me). Hopefully. 
2. Developers are welcome to join freely. pull requests and issues are welcome there.
2. Although I'm not familiar with all the simulator details. I will try to modify the latest release to let it run standalone, independent of the WISE cloud service which will probably go offline in the middle of the year.
3. Currently I'm fighting alone and I'm looking forward to more developers joining :)

rongguodong, 2022-01-25:
It is a great start of the new era of SVL Simulator. Thanks a lot for initiating this!

SVL Simulator has grown into one of the best open source AV simulators. Particularly, it natively supports the two mainstream open source AV platforms, Apollo and Autoware, which makes it much easier to use than other similar simulators. Really hope this project can continue to grow in the future.

daohu527, 2022-01-26:
Open source is like opening a museum, people can come and observe, but very few people actually create it.

Create what you can, great things attract people : )

hadiTab, 2022-01-26:
@ntutangyun the commit history for the simulator repo seems to have been wiped. I'd suggest preserving it as it would make future modifications much easier.

ntutangyun, 2022-01-26:
Hi @hadiTab I was about the import the repo instead of starting from scratch. But the large file storage quota for free organizations is limited (1GB), insufficient for the LGSVL repo ( >4GB). Thus I could only add all the files back to the normal git and commit it all together for the time being :)

tuanngokien, 2022-04-26:
Hi @ntutangyun,
Are you having any near plan that tending to remove WISE dependencies from LGSVL?
I have tried #1994 and it worked quite well on Python API mode, I think you could try it as well.

closed 2022-08-11 by ntutangyun