lgsvl/simulator #1970: 320	../sysdeps/nptl/futex-internal.h: 没有那个文件或目录(no such file or directory)

jackxia123, 2022-01-26:
i just run gdb ./simulator, it threw below error, how to solve this problem

New Thread 0x7fff0c683700 (LWP 2747472)]
--Type <RET> for more, q to quit, c to continue without paging--

Thread 39 "Finalizer" received signal SIGPWR, Power fail/restart.
[Switching to Thread 0x7ffff18eb700 (LWP 2747434)]
futex_abstimed_wait_cancelable (private=0, abstime=0x0, clockid=0, expected=0, futex_word=0x7ffff2397880)
    at ../sysdeps/nptl/futex-internal.h:320
320	../sysdeps/nptl/futex-internal.h: 没有那个文件或目录(no such file or directory)

EricBoiseLGSVL, 2022-01-29:
Sounds like this is a hardware issue.  What GPU do you have?

jackxia123, 2022-01-29:
A40

发自我的iPhone

EricBoiseLGSVL, 2022-01-31:
Unity applications are not meant to run on this GPU without some work.  You will need to research how to run on a server with this hardware.