lgsvl/simulator #1971: New cluster is detected offline while it is online?

momala, 2022-01-26:
Hi everyone,

After hearing the shocking news from the SVL team, I faced a problem that was weird. The simulator does not run any simulation with new clusters that I have created today. Although in the cluster tab, it shows me that the cluster is Online but in the simulations tab the simulation item shows me the cluster is offline and does not allow me to run any simulation. I also tested with the default map and vehicle but nothing changed.
Do you have any ideas? is it related to the SVL team's recent decision? 

This is my simulation Tab: [Cluster is not available for running simulations]
[image]
This is my Cluster Tab:
[image]

sethut, 2022-01-26:
Did you solve it? I have the same problem from today
I thought it was just my problem, but it doesn't seem to be

momala, 2022-01-27:
> Did you solve it? I have the same problem from today I thought it was just my problem, but it doesn't seem to be

Unfortunately not. I don't know what happened. If anyone has already a cluster don't touch it! it works. But if you remove it you get into trouble!

working with Wise seems not wise anymore!

brkay54, 2022-01-27:
It is not permanent solution but if you have any cluster you created before, you can add the device in that cluster then make it master. After that remove the device you added before. It works for me.

momala, 2022-01-27:
> It is not permanent solution but if you have any cluster you created before, you can add the device in that cluster then make it master. After that remove the device you added before. It works for me.

Thanks for mentioning it. I removed my previous clusters in the debugging process. I don't have any chance.

simongohtj, 2022-01-27:
I'm also facing the same problem. New clusters are no longer working.

thisisdavidfrank, 2022-01-27:
@momala @simongohtj @sethut, sorry about this. We are working on a fix. Will update soon.

eagafonovlge, 2022-01-27:
@momala @simongohtj @sethut @brkay54 

We have deployed a hotfix update for the issue. 
You need to delete the troubled cluster, create a new one, and update affected simulations.

Please confirm if it fixes the issue for you.

closed 2022-01-27 by eagafonovlge

momala, 2022-01-28:
> @momala @simongohtj @sethut @brkay54
> 
> We have deployed a hotfix update for the issue. You need to delete the troubled cluster, create a new one, and update affected simulations.
> 
> Please confirm if it fixes the issue for you.

Yes, It is working now. Thanks