lgsvl/simulator #1972: Is the wise.svlsimulator.com closed after June 30 ?

poco919, 2022-01-27:
Hi
First, thank you for the team for deceloping this great simulator.
I'm so sad to hear the end of development.

I would like to confirm one thing.
The document says 'We will make a reasonable effort to keep the wise.svlsimulator.com website up and running through at least Thursday, June 30, 2022, and will also maintain the SVLSimulator.com web site including the simulator documentation pages through at least Thursday, June 30, 2022.'.
Dose it mean the web site will be closed after June 30 and we need to make some modification to continue to use on local environments?
Or, is the web site stilln hosted after June 30 without supporting?

Thanks

EricBoiseLGSVL, 2022-01-29:
The site will be closed.  You will need to edit the simulator to use local assets by this date.  See Developer Settings code in simulator, it will show how to use local assets.

poco919, 2022-01-31:
Thanks. I will try to use local assets.

closed 2022-01-31 by poco919

rogeherlor, 2022-03-11:
> The site will be closed. You will need to edit the simulator to use local assets by this date. See Developer Settings code in simulator, it will show how to use local assets.

Hi
Did you refer to DeveloperDebugModeToggle, DevelopmentSettingsAsset and DevelopmentSettingsWindow scripts? Where is it shown?