lgsvl/simulator #1973: Facing error while running simulation based on visual scenario editor

momala, 2022-01-28:
Hello,

I was trying to use the Visual editor feature to create a scenario. I created a simulation by choosing the test case as a visual scenario editor. I imported the Scenario JSON file. But when I run the simulation it gives me the error below: 

`Error on latest run: Test case exception: System.Net.Http.HttpRequestException: Connection failed ---> System.Net.Sockets.SocketException: Connection refused at System.Net.Sockets.SocketAsyncResult.CheckIfThrowDelayedException () [0x00014] in <aa976c2104104b7ca9e1785715722c9d>:0 at System.Net.Sockets.Socket.EndConnect (System.IAsyncResult asyncResult) [0x0002c] in <aa976c2104104b7ca9e1785715722c9d>:0 at System.Net.Sockets.SocketTaskExtensions+<>c.<ConnectAsync>b__2_1 (System.IAsyncResult asyncResult) [0x00006] in <aa976c2104104b7ca9e1785715722c9d>:0 at System.Threading.Tasks.TaskFactory`1[TResult].FromAsyncCoreLogic (System.IAsyncResult iar, System.Func`2[T,TResult] endFunction, System.Action`1[T] endAction, System.Threading.Tasks.Task`1[TResult] promise, System.Boolean requiresSynchronization) [0x00019] in`

I don't have any idea what kind of error is. But I would be thankful if anyone can tell me if they can use this feature(Using the visual scenario creation and running a simulation based on that)? Or maybe it is related to the project's sunset?

EricBoiseLGSVL, 2022-01-29:
Scenario Editor requires Apollo dream view running, do you have the cyber bridge up before you run the scenario?

momala, 2022-01-30:
> Scenario Editor requires Apollo dream view running, do you have the cyber bridge up before you run the scenario?

Thanks Eric for the reply. No, I don't have it. Do you mean this feature just works for the Apollo or this is a basic requirement? Because I am going to use it with Autoware.

EricBoiseLGSVL, 2022-01-31:
The runner requires Apollo, it doesn't support Autoware at this time.  We wanted to release a version that will support any but we weren't able to release before the project was sunset.

momala, 2022-02-01:
> The runner requires Apollo, it doesn't support Autoware at this time. We wanted to release a version that will support any but we weren't able to release before the project was sunset.

Thank Eric for your support.