lgsvl/simulator #1974: Possible bug in GPS sensors when running SVL with Autoware.AI

ZiwenWan, 2022-01-31:
Dear LGSVL team, 

Thanks for the efforts in creating the simulator! I recently met some problem when launching SVL(2021.2) with Autoware.AI (version 1.14). I think there might be some bug in GPS sensor or bridge and leads to incorrect position/heading. Here is an example on Highway 101 map. The first picture shows the position of ego vehicle in the simulator and the second picture shows the position of Autoware's location. 
[image]
[image]

I have also tried SVL 2021.3. The position can match correctly, but somehow there is still some error in the heading information. Could you please share some information about how to fix this (i.e., share some possible transformation I should apply to fix the error)?

Thanks a lot!

EricBoiseLGSVL, 2022-02-01:
You can look at the sensor changes from 2021.2 to 2021.3 to see what is changed.  I would also look to the bridge to see any conversions that were changed.  In the sensor you should see an option to ignore map origin.

ZiwenWan, 2022-02-02:
Thanks a lot for your comment! I will take a look at that. I think this might be related to the map origin (the center point should be the same while there is some error in rotation).