lgsvl/simulator #1977: Some element in the map are not loaded by Python API

momala, 2022-02-01:
I am trying to load a pre-built map into the simulations. While I am using the Wise to create a simulation, It runs and shows the map without a problem But when I use the python API to select that map as the environment, it doesn't show the terrains (vegetation and other objects). [NB: We have a ground mesh but the details are off ground and utilizes the builtin point cloud renderer of svl simulator]
Picture : scene loaded with Wise
[image]
Picture: Scene loaded by API
[image]

I am using these line to load the map.
`if sim.current_scene == lgsvl.wise.DefaultAssets.map_taltech:
    sim.reset()
else:
    sim.load(lgsvl.wise.DefaultAssets.map_taltech)`
Do you have any Ideas?

EricBoiseLGSVL, 2022-02-01:
That seems to be a bug.  It looks like the api code is not loading the pointcloud data.  Look to the loading code for wise in loader and grep the difference in the api command load()

momala, 2022-02-02:
> Look to the loading code for wise in loader

Thanks for the reply. I couldn't find exactly where the scene loading happens in wise. I am looking this two file:
Web: simulator/Assets/Scripts/Web/Loader.cs
API: simulator/Assets/Scripts/Api/Commands/LoadScene.cs

But I couldn't find where exactly it loads the scene in the web. Could you please give me some clue?

EricBoiseLGSVL, 2022-02-02:
Loader.cs has the method StartAsync(SimulationData simulation).  This is the way random traffic maps are loaded.
LoadScene.cs is where API maps are loaded.

Internally we moved all loading to API commands but we weren't able to release before sunsetting simulator, sorry.

momala, 2022-02-07:
> Loader.cs has the method StartAsync(SimulationData simulation). This is the way random traffic maps are loaded. LoadScene.cs is where API maps are loaded.
> 
> Internally we moved all loading to API commands but we weren't able to release before sunsetting simulator, sorry.

Thanks again Eric for your help.

closed 2022-02-07 by momala