lgsvl/simulator #1979: Is the random traffic script open source for users?

JasonBourne1998, 2022-02-03:
When my classmates and I build a scene in the pythonAPI and connect with Apollo's dreamview, activating prediction, performance and traffic light will occupy a lot of GPU memory. However, when we use random traffic to control the car and connect it with dreamiew, it will be much easier to load the model. Therefore, we want to modify the scene directly on the basis of random traffic. Is the script of random traffic open source?

lemketron, 2022-02-03:
Random Traffic was implemented directly in SVL Simulator, but recently we implemented it using Python API. 

You can find the Python script for Random Traffic in the (new) "runner" repo here: 
https://github.com/lgsvl/runner/blob/master/random-traffic-runner/scenario_runner/run_random_traffic.py

Note that this was implemented with a specific future use case in mind and may or may not work as-is with SVL Simulator 2021.3 (the latest public release).

JasonBourne1998, 2022-02-08:
> 

Thanks for your reply, now we have run it successfully, but now we want to join pedestrians who are crossing the road, but it seems that lgsvl does not provide the insertion of pedestrians who cross the road obey the traffic rules. Now it only supports walking randomly on the crosswalk and walking according to the waypoint. However, as we all know, when pedestrians cross the road, they need to cross according to the red street lights. Does lgsvl have a script in this regard? If not, I wonder if I can use randomly inserted NPC cars to transform the script to create pedestrians who comply with traffic regulations? (because the cars randomly inserted in lgsvl obey the traffic rules), is this feasible?

JasonBourne1998, 2022-02-08:
> Random Traffic was implemented directly in SVL Simulator, but recently we implemented it using Python API.
>
> You can find the Python script for Random Traffic in the (new) "runner" repo here: [...]

Thanks for your reply, now we have run it successfully, but now we want to join pedestrians who are crossing the road, but it seems that lgsvl does not provide the insertion of pedestrians who cross the road obey the traffic rules. Now it only supports walking randomly on the crosswalk and walking according to the waypoint. However, as we all know, when pedestrians cross the road, they need to cross according to the red street lights. Does lgsvl have a script in this regard? If not, I wonder if I can use randomly inserted NPC cars to transform the script to create pedestrians who comply with traffic regulations? (because the cars randomly inserted in lgsvl obey the traffic rules), is this feasible?

EricBoiseLGSVL, 2022-02-10:
Yes, you will need to alter pedestrian control classes and may need to add pedestrian annotation data (stoplines, etc).

EricBoiseLGSVL, 2022-02-10:
Yes, Borregas Ave is not open source.  If you can't make a map to edit, then you will need to use the annotation provided.  Have you used the VSE editor?  Maybe you can annotate paths and then edit the api pathing to randomize and follow pedestrian rules.  The best bet is to make annotation edits to a new map and create new annotation data/logic to succeed.