lgsvl/simulator #1981: New ego vehicles upload to wise fails

farukbaykara, 2022-02-08:
I'm trying to build a new ego vehicle, and the build is successfully completed in the unity editor. While uploading to the wise I get the following error
[image]
Size of my AssetBundle is really big (550Mb). Can this cause this problem to occur? If this is the problem, how can I reduce the size of my AssetBundle?

StepTurtle, 2022-02-09:
i had same problem,  and it was because my model was big over than 200 mb. Wise does not allow files over 200mb. You have to decrease your model size.

EricBoiseLGSVL, 2022-02-10:
Yes, you need to remove texture data from the fbx (import separately in Unity) and reduce poly count of mesh