lgsvl/simulator #1982: New ego vehicle not moving

StepTurtle, 2022-02-09:
Hello, I uploaded my own vehicle model to SVL but I can't control it, it is not moving. When I press the back button, the brake light turns on, so there is no problem with the keyboard layout. Is there any solution for this.
[image]

EricBoiseLGSVL, 2022-02-09:
did you add a keyboard control sensor to the vehicle config?

StepTurtle, 2022-02-10:
[image]
![WhatsApp Image 2022-02-10 at 13 41 00](https://user-images.githubusercontent.com/76053179/153392945-ac449269-9464-4d7
[image]
e-9c4f-16bcdc2ed0ed.jpeg)

Are the pictures I sent enough?

StepTurtle, 2022-02-15:
Sorry for my late feedback, some ill problems. I changed the aligning. I assign the correct colliders to the wheels. My car is moving. Thanks to you. Have a good day.

closed 2022-02-15 by StepTurtle