lgsvl/simulator #1986: The navmesh of sanFrancisco

JasonBourne1998, 2022-02-15:
We imported sanfrancisco map in unity, but now we can't use the navmesh of the map. Is this the problem of the map itself? And do we need to design that navmesh by ourselves?

EricBoiseLGSVL, 2022-02-16:
What do you mean imported?  Did you clone the repo?

EricBoiseLGSVL, 2022-02-18:
Walk randomly is a api command for pedestrian waypoint mode behavior.  Open the Unity editor and run an api sim with this command.  Then select and pedestrian and look to the behavior assigned at runtime.  The api command classes have this information on the method called.

xli149, 2022-02-22:
> Walk randomly is a api command for pedestrian waypoint mode behavior. Open the Unity editor and run an api sim with this command. Then select and pedestrian and look to the behavior assigned at runtime. The api command classes have this information on the method called.

Hi Eric. I'm facing the same problem of tracing the way points of pedestrians with running walk_random api in simulator. The problem is that I need to collect the way points from the simulator directly since I have to connect the simulator to Dreamview and need to record the trajectories of each pedestrians in real time. Do you know if there is a way that I can achieve this from the simulator api?

Best Regards