lgsvl/simulator #1987: Import HD map Lanelet2 fail? InvalidOperationExecption: Nullable object must have a value.

llu0120, 2022-02-16:
I tried to use the lgsvl simulator and I'm using the `tag 2021.3` in the `simulator` repo. However, when I'm trying to import my lanelet2 map into the simulator, I can't see my map inside unity and report an error below. Has anyone confront this before and can help with this? Thanks in advance. 
 
[image]

llu0120, 2022-02-21:
Hi @EricBoiseLGSVL, 
Thanks for your reply. I've tried to import the example lanelet2 map from BorregasAve, but the import visualization looks strange as well. 
[image]

Looks like only the traffic lights are being imported. Is it normal?

EricBoiseLGSVL, 2022-02-23:
Do you have generate mesh setting enabled in the import settings?

llu0120, 2022-02-24:
no, but I can see the lanes when clicking view all in the HD Map Annotation mode. Is that normal? Thanks!

EricBoiseLGSVL, 2022-02-24:
Yes, this is just annotation data, not a drive-able mesh.