lgsvl/simulator #1988: Pedestrian crosswalk trajectory does not fit the path in some directions

xli149, 2022-02-17:
I have followed the Map annotation guide to build the pedestrian lane in order for the pedestrians to obey the traffic rules. The pedestrians' trajectory fit the pedestrian lanes on the other two direction of the crosswalk. However, they fails to fit in the lane in this direction as following. By not fitting in lane, I mean, the direction is correct, but there is a huge deviation between the pedestrian lane and the actual pedestrian trajectories no matter how I adjust the lane. Have some one encountered with this problem before?
[image]

EricBoiseLGSVL, 2022-02-18:
Yes, random traffic has randomization in the path and will recalculate often during runtime.  If you need precision in the path, use the api to specify the path.

xli149, 2022-02-20:
Thank you for your explanation Eric. However, this seems still a little bit weird.  The pedestrians on the other two lanes are walking within the boundary, but tend to walk outside the boundary in this lane as following. Is there any reason for this?
[image]
[image]
[image]
?