lgsvl/simulator #1991: Dreamview and lgsvl's San Francisco map directions are inconsistent

xli149, 2022-02-20:
Hi, I've tried to put the bin files of the map SanFrancisco in to the /apollo/modules/map/data and the Dreamview can successfully show the map when I run the random traffic in lgsvl simulator. However, the only problem is that there seems to have an orientation mismatch between the two map orientations. Is there a way fix this issue?
[image]

ntutangyun, 2022-02-20:
download the map file from here https://content.lgsvlsimulator.com/maps/sanfrancisco/

and replace the maps in the apollo map folder and try again ?

xli149, 2022-02-20:
> download the map file from here https://content.lgsvlsimulator.com/maps/sanfrancisco/
> 
> and replace the maps in the apollo map folder and try again ?

Thank you ntutangyun, I have tried. Those are the map files I have included but the problem still exist. I tried to look into the .txt files to find some information (maybe related to rotation? I don't know..) but I haven't find any clue yet. The strange thing is that cubetown  seems works perfectly which does not have this kind of mismatch when I import the related map files into apollo/module/map/data
[image]
.

ntutangyun, 2022-02-20:
after you updated the map file, have you restarted the dreamview ?

xli149, 2022-02-20:
> after you updated the map file, have you restarted the dreamview ?

Yes, thanks for reminding me. I actually stopped and restarted the docker and re-runned the boostrap_lgsvl to make sure the caches have been cleaned. Sadly the problem still exist.

ntutangyun, 2022-02-20:
how about download the assetbundle of the san Francisco map from the link above and load it into the simulator ?

xli149, 2022-02-22:
> how about download the assetbundle of the san Francisco map from the link above and load it into the simulator ?

Hi, ntutangyun, sorry for the late reply. I've tried to upload the map assetbundle to simulator but failed with this error.

[image]

MINGG2333, 2022-03-21:
Hi, I meet the same problem, have you solved it?
I also have tried another way to add the HD map of SanFrancisco according to https://github.com/ApolloAuto/apollo/tree/master/modules/map/data using `base_map.bin` from https://wise.svlsimulator.com/maps/profile/5d272540-f689-4355-83c7-03bf11b6865f, but the problem still exists.

I don't know how to upload the map asset bundle to the simulator, how did you do that?
I use the master branch of Apollo and the latest svl simulator (linux64-2021.3).
[image]

Addi2020, 2022-04-21:
@ntutangyun 
I use Apollo7.0 and svlsimulator-linux64-2021.3.

I have same issue with SF map and get error while uploading Asset bundle.

yang-yk, 2022-12-28:
@MINGG2333, hello, have you solved this problem with the SF map? Could you please share the solutions with me? Wish to hear from you soon!