lgsvl/simulator #1992: visual editor win32exception

wannastar, 2022-02-21:
i used svl2021.03 simulator visual editor to create my own scenario, but when i finished my scenario and to simulator, i cannot run my own scenerio. informed win32exception(0x80004005):系统找不到指定文件
[image]

EricBoiseLGSVL, 2022-02-24:
Do you have docker for windows installed as per our docs [here](https://www.svlsimulator.com/docs/installation-guide/installing-simulator/#docker)?

wannastar, 2022-02-25:
oh, sorry, i forget this... but after i installed docker desktop and wsl，when i run my own scenariro created by VSE, it started downloading svklcontent/runner/vse and then informed  Error unaothorized authentication required. i wanna ask where to input the sign in info or how to resolve this question
[image]

EricBoiseLGSVL, 2022-02-25:
are you authenticated on docker for windows?

wannastar, 2022-03-01:
yeah. Thanks for your advice. I had resolved my problems.

closed 2022-03-01 by EricBoiseLGSVL