lgsvl/simulator #1993: Segmentation shader in the HD rendering

yumianhuli2, 2022-02-21:
Hello!
In the LG simulator, is there a shader suitable for semantic segmentation in the HD rendering mode？
Thank you!

EricBoiseLGSVL, 2022-02-24:
HDRP and URP are very different shader wise.  We provide segmentation sensor using edits to HDRP for the Lit shader in Unity [here](https://www.svlsimulator.com/docs/simulation-content/sensors-list/#segmentation-camera).  Is this what you mean?

yumianhuli2, 2022-02-25:
> 

Yes, Could you plz share all the relative code about semantic segmentation in lgsvl?
Thank you!

EricBoiseLGSVL, 2022-02-25:
Sensor [here](https://github.com/lgsvl/SegmentationCameraSensor)
Simulator has code to work with the sensor in HDRP and shaders [here](https://github.com/lgsvl/simulator)

closed 2022-02-25 by EricBoiseLGSVL

yumianhuli2, 2022-03-02:
> Sensor [here](https://github.com/lgsvl/SegmentationCameraSensor) Simulator has code to work with the sensor in HDRP and shaders [here](https://github.com/lgsvl/simulator)

Thank U！