lgsvl/simulator #1994: OSSDC SVL based simulator which doesn't require SVL cloud (wise.svlsimulator.com)

mslavescu, 2022-02-21:
Here is my initial version of SVL simulator that can be used without wise.svlsimulator.com:
https://github.com/OSSDC/simulator/tree/release-2021.3

My plan is to continue the development of SVL Simulator as part of http://OSSDC.org project.
If you'll like to help, join OSSDC Slack at http://join.ossdc.org

A Windows build preset with a few simulations is available here:
https://race.ossdc.org/simulator/simulator-ossdc-20220221.zip

To easily test this build follow these steps::

- unzip simulator-ossdc-20220221.zip
- start OSSDC Simulator with run-svl-2021_3-no-svl-cloud.bat
- launch OSSDC API simulation (already preset)
- clone https://github.com/OSSDC/F1TENTH_SVL-Simulator
- run one of these API based simulations scenarios:
F1TENTH_SVL-Simulator/Python-Scripts$ python3 02_Pure_Pursuit.py
F1TENTH_SVL-Simulator/Python-Scripts$ python3 03_Multi_Ego.py

I'll add later today a video with a sample run 
In next few days I'll add more scripts to run with lgsvl_bridge inside AutowareAuto ADE env (like in my first video in the playlist bellow).

See more of my SVL test videos in this playlist:
[SVL simulator racing tests](https://www.youtube.com/playlist?list=PLUop7b1Q1uZmYR5WK5XegykPhb9fZNtPW)

JOLOC, 2022-02-23:
Hi, I would like to know how to use this simulator you proposed, can you update an explanation video on your twitter if possible

tuanngokien, 2022-04-26:
Thank you @mslavescu for modifying the simulator so now we don't need to worry much about LGSVL turning down the WISE cloud. Also, in the original post, the author only shared the built preset for Windows, but last time I saw that many people using this simulator were working on Ubuntu too. Thus, I have built @mslavescu's release on Ubuntu 18.04 and want to share the binary with you guys.
#### Download link: https://drive.google.com/file/d/1FjZSuV3MC8uqi8mvoyZVeN3YwkBwu31o/

#### Running instruction:

1. Unzip `linux-simulator-ossdc-20220426.zip` to folder `linux-simulator-ossdc-20220426` 
2. Change directory `cd linux-simulator-ossdc-20220426`
3. Because the simulator has been built, you only need to open terminal and run `./run_lgsvl_without_wise.sh` (as Figure 1)
4. Choose config `OSSDC-API` and click `Start` to have LGSVL run on Python API mode''
5. To run a demo, move inside folder `./F1TENTH_SVL-Simulator-main`
                    5.1. Install LGSVL Python API dependencies `pip3 install --user -e lgsvl-PythonAPI-master/.`
                    5.2. Run demo script `python3 Python-Scripts/01_Spawn_and_drive.py`
                    5.3. LGSVL now running the scenario (as Figure 2)

Figure 1             |  Figure 2
:-------------------------:|:-------------------------:
[image]  |  [image]

lintglitch, 2022-04-28:
Thank you so much for the work! Do you know if there is a work-around for the _Development Settings_ window when opening the simulator in Unity? With that new scenarios and sensor configuration can be defined.

The _Development Settings_ window always checks for cloud connectivity first and will not open without the WISE server. I've tested it with this fork and the issue remains. If we could disable the check, we could still define new scenarios and test-vehicles after the sunset.

tuanngokien, 2022-05-06:
Hi @lintglitch, 
I took a look at _Development Setting_ in Unity and seems that it's similar to which we have on WISE while creating new simulation. I think I could have idea for your issue, but I need more time to work on it.

Now from Unity, assuming that we could pass WISE connectivity check and have all options for a new simulation config. My question is how we could export this config for using it in simulator, because I don't see any SAVE button in _Development Setting_ popup. If you have done that, could you record a video telling step-by-step that I can follow to create a scenario from Unity?

lintglitch, 2022-05-13:
Hey @tuanngokien,
sorry for the late reply. Not sure if you could export the development settings, but I think it should be possible in principal.

The idea behind getting the Development Settings to work is that you can then define any scenario directly within Unity and run them from there. I have used that feature extensively for development of new maps and vehicles, since it allowed me to test them without exporting and uploading them first. If that interests you I can upload a video of my step-to-step to get that working.

Essentially if we can bypass the Development settings check we can create any new map, vehicle or sensor configuration we like. However, the disadvantage is that that approach only works from within Unity.

tuanngokien, 2022-05-13:
Got it, @lintglitch! 
Hope you could further give me the video. Thank you!

lintglitch, 2022-05-16:
No problem, here is the video @tuanngokien.

https://user-images.githubusercontent.com/2789099/168600864-92835f0b-4756-4836-9afd-670ff2f7a603.mov

The feature itself is not well (or at all) documented, but works quite well. Here are the steps I took:

1. Open a map inside Unity. Note that I haven't uploaded this map or exported it as an assetbundle. It is simply not necessary.
2. Open the development settings and select the ego-vehicle and sensor configuration you want. Significantly, this step fails when not connected to the cloud. I have selected a custom ego-vehicle that was also not uploaded to the cloud or exported as an asset bundle
3. Hit play and enjoy.

Everything else works as usual inside SVL. I have been using this approach for some time to accelerate development of new cars and maps. It also has the advantage of allowing you to edit the simulation while it is running, a nice functionality for testing.

Hope the video helps you. Thanks for the good work!

mslavescu, 2022-06-02:
Here is the OSSDC SIM, based on LG SVL Simulator, to further evolve this amazing project:
https://github.com/OSSDC/OSSDC-SIM

Will provide demos shortly.

josh-wende, 2022-06-02:
@mslavescu great, thanks! I've kinda figured out how most of it works now, I'm currently just struggling to get it to connect to a locally-built Cyber bridge. Definitely going to keep up with the OSSDC project!

emocat, 2022-06-06:
@josh-wende Hi, I'm currently also struggling to get it to connect to a locally-built Cyber bridge, but don't have any clue now. Could you please share your solutions on how to add assets and construct sensor configuration locally? Thanks a lot!

josh-wende, 2022-06-06:
@emocat I was able to figure it out by following the template of the assets included in @tuanngokien's pre-built linux example. The AssetBundles go in the data folder (`2021.3-ossdc` in this example case), and you need to manually add entries for each new asset to `data.db`, which is in the data folder as well. Then the simulator looks locally for a JSON that defines the entire map/vehicle. In this example setup it looks for the JSON in `wise/api/v1`. Follow the template of the JSONs already in there for the map and vehicle, replacing asset GUIDs with the ones from your `data.db`. The only change to the template I needed was to move the `bridgePluginId` an `bridge` fields out of the `sensorsConfigurations` field and into the top level of the JSON. Hope this helps! (Also feel free to ask this same question at https://github.com/OSSDC/OSSDC-SIM/issues and I will copy my response over there)

mslavescu, 2022-06-19:
We just posted first OSSDC SIM demo video and simulation instructions, see details here:
https://github.com/OSSDC/OSSDC-SIM-Demos/discussions/2

mslavescu, 2022-06-26:
Check this comment on how to use other maps and vehicles configs in OSSDC SIM:

https://github.com/OSSDC/OSSDC-SIM/issues/3#issuecomment-1166593428

I also cloned the https://www.svlsimulator.com/docs here:

https://race.ossdc.org/docs/

Search doesn't work but all pages should be there.

Sam827-r, 2022-11-29:
Hi @josh-wende @tuanngokien 

I am not getting how Python API to load assets from local files, for while able launch simulator using script ./ run_lgsvl_without_wise.sh
i have gone through both binaries provided by @mslavescu @tuanngokien  but no luck. 
working on Ubuntu 20.04