lgsvl/simulator #1996: Custom map on the visual scenario editor?

lakshjaisinghani, 2022-02-24:
I was wondering if we can add custom maps to the visual scenario editor (VSE). I'm simulating near misses and that's not really possible on existing maps where vehicles "lock onto lanes". 

I was also wondering where these lane behaviors were defined, for example in the SanFransisco Map. This is because I can spawn waypoints at any coordinates in the VSE in a parking lot, but that is not possible on roads.

lakshjaisinghani, 2022-02-24:
One can create a SanFransisco map without "Lane Snaps" by removing all but one **_TrafficLane_** components in the SanFransisco (SF) map.

I have already done this, all you need to do is download [this map](https://wise.svlsimulator.com/maps/profile/1f903cb2-477e-44aa-a8c0-a29af6d5e433) from the store and use it. This replaces the original SF Map.

EricBoiseLGSVL, 2022-02-24:
You can create any map you want.  If you want to specify waypoints, then use VSE and don't annotate the map on the section you want to create scenarios.  There might be a bug in maps without annotation and using the API for VSE maps.