lgsvl/simulator #2005: The question about the environment panel in the simulator

JasonBourne1998, 2022-03-01:
Why the UI interface can't display the interactive window of the environment panel in the bottom corner when the simulator is simulated with PythonAPI, but there is this window when we directly use the random traffic mode in the website?
The random traffic:
[image]

The PythonAPI:
[image]

JasonBourne1998, 2022-03-01:
Besides, we think that this problem caused by the interactive mode, in the pythonAPI, we could not open the interactive mode.

carTloyal123, 2022-03-25:
With the PythonAPI you can set all of the needed parameters for the environment so you should not need the menu at all. Check the examples here: https://www.svlsimulator.com/docs/python-api/python-api/#quickstart