lgsvl/simulator #2006: No SpawnInfo generated when I enter a new Scene

llu0120, 2022-03-01:
Hi, 
I confront a problem for no SpawnInfo automatically generated when I open a new scene. 
This is how I open a new scene.
[image]

and I choose the Basic (Built-in) one 
[image]

And no SpawnInfo is included. Did anyone confront this before? 

Thanks in advance.

jim18324, 2022-03-02:
I think you should choose Simulator Environment. Or simply copy and paste from an existing scene and modify it by yourself.

llu0120, 2022-03-03:
Thanks @jim18324 !

closed 2022-03-03 by llu0120