lgsvl/simulator #2010: Opening the simulator as a project in Unity

YigithanGediz, 2022-03-03:
Hi all!

I want to open the simulator in Unity to add a new map as indicated in official documentations. However, I get compilation errors and "Simulator" option does not show in the dropdown menu. I am opening it with 2020.3.3f editor with Linux - Windows build support. My OS is Windows 10

After struggling for a while, I tried to follow the steps in here:, https://www.svlsimulator.com/docs/archive/2020.06/build-instructions/ 

The problem was the step below:
10) Open a terminal and navigate to the WebUI folder of the Simulator project

There is no WebUI folder in the Simulator project?

Can someone help me please?

Thank you in advance!

jim18324, 2022-03-07:
maybe you forget to switch git branch to release-2020.06?

Miknow78, 2022-03-09:
> Hi all!
>
> I want to open the simulator in Unity to add a new map as indicated in official documentations. However, I get compilation errors and "Simulator" option does not show in the dropdown menu. [...]

Did you find the folder?

rogeherlor, 2022-03-10:
I am not sure but I think you had to skip 10. , do 11. and 12. and then the folder was generated

YigithanGediz, 2022-03-13:
@jim18324 Thank you! I had to clone release branch 2021.3 of lgsvl simulator repository.

@Miknow78 Actually there is no need for that. We have been looking to old lgsvl documentations. Correct documentation is here
https://www.svlsimulator.com/docs/installation-guide/build-instructions/