lgsvl/simulator #2013: Ground Truth can' t detect the objects

buseerdem, 2022-03-04:
I upload ground truth to my simulator and i am trying to get ground truth data via ros. I also added ground truth 2d and 3d sensors in wise. But i can' t detect anything as you can see from the photos. How can i get any data from ground truth ?
[image]
[image]
[image]

wannastar, 2022-03-09:
maybe you can solve it by changing the sensor's positon, such as y = 1.4

wannastar, 2022-03-10:
修改位置是为了传感器不存在遮挡。
你可以看着传感器自身坐标系调节，修改的应该是Z轴，设置z = 1.5
我之前是通过这种方式解决的这个问题。

rogeherlor, 2022-03-10:
Did you specified the ROS topic in wise?