lgsvl/simulator #2016: Highway101GLE map reset doesn't work

Miknow78, 2022-03-07:
Hi!
I'm using Highway101GLE map.

if sim.current_scene == "431292c2-f6f6-4f5a-ae62-0964f6018d20":   
    sim.reset()
else:
    sim.load("431292c2-f6f6-4f5a-ae62-0964f6018d20")

At the first time I run the script, simulation works fine.
But at the second time, reset does not work and I have to stop simulation in the browser and start again.
Any ideas?

emocat, 2022-03-08:
Maybe there is a conflict between the api's you call? 
As for me, when I accidentally call reset() twice in a row, or call reset() after sim.close(), the simulator gets stuck. So maybe you should check your script more carefully.

Miknow78, 2022-03-08:
Thank you for your reply!
My code is like this:

if sim.current_scene == "431292c2-f6f6-4f5a-ae62-0964f6018d20":
sim.reset()
else:
sim.load("431292c2-f6f6-4f5a-ae62-0964f6018d20")

Then, I start the simulation with sim.run(5) and sim.stop().

Can you please send me your code structure for me to try to understand?
Thank you so much.

emocat, 2022-03-08:
Well, I don't find anything wrong in your code. For me, I usually call reset() and close() after a simulation is finished.

Maybe you can check lgsvl's log to find more useful information.

Miknow78, 2022-03-08:
Sorry, but I don't know who to ask.
Can you please explain me how to add a plugin to my simulation?

I just want to know what type of line my vehicle has on your right/left.
I saw the Lane-line sensor plugin. But I don't know how to implement it in
my simulation.

emocat, 2022-03-08:
If this plugin is published on https://wise.svlsimulator.com, you can directly add it to your vehicle's sensor configuration. Or you need to compile the sensor by yourself and publish it.

Miknow78, 2022-03-08:
I figured it out how to call the sensor configuration in my simulaton :)

Thank you so much for your help!