lgsvl/simulator #2017: the resources of sensor map and so on is always doanloaded again when i changed my PC. how can i use my offline? resources

wannastar, 2022-03-08:
the resources of sensor map and so on is always doanloaded again when i changed my PC. how can i use my offline resources?

jim18324, 2022-03-09:
maybe build from source and using developer debug mode?

carTloyal123, 2022-03-25:
You should be able to use your own local resources by going to your persistent directory on the first machine. Copy all of the contents and then paste them into the persistent directory of the second machine. 

For windows: C:\Users\<YOUR USERNAME>\AppData\LocalLow\LGElectronics\SVLSimulator-<VERSION>
For linux: $HOME/.config/unity3d/LGElectronics/SVLSimulator-<VERSION>

Note that you will not be able to create a new scenario without using the web interface. Hope this helps.